kakopappa/sinric #282: Sending back the status to google home

maciak, 2019-05-19:
Hi,
I have a question about sending the status of colour light fo GH. With just turning it on and off it works fine but for example, when I turn the light straight to some color on main light screen it shows off. The same if I as the speaker to set light to red for example it's the same I open google home app go to the light and it says off.

Another thing is with a thermostat. It's always set to cool. When I try to change it to heat or off it tries to update and coed back to cool.

Is there any way of updating those statuses so they look fine in the GH app?

kakopappa, 2019-05-21:
Sorry about the late response. it was long weekend where i live. 

1. So to reproduce this issue:

1. Turn on light. 
2. Change the color
3. Check the light status on app 

is this correct ?

2. I have to take a look at the code.

maciak, 2019-05-23:
Hi,
No to reproduce turn the color light by voice. Example: Turn light to red.
Check the app and it says off.

kakopappa, 2019-05-23:
Ok, I will take a look at this weekend

maciak, 2019-05-29:
HI,
Any progress on this?

kakopappa, 2019-05-30:
sorry not yet.