kakopappa/sinric #283: How to trigger an Amazon/Google routine with the Button example?

temocastro, 2019-05-19:
I am trying to make a routine with Amazon, but I cant see the button as an option to trigger a routine. 

Been reading, if the button could be connected to SmartThings, I could use it.

Does anybody have an idea?

temocastro, 2019-05-24:
Been reading and it seems the device Sinric needs to be it's a Motion Sensor, there is no Motion Sensor device option in Sinric @kakopappa could you add one?

temocastro, 2019-06-01:
@kakopappa help

kakopappa, 2019-06-01:
hi, sorry I missed your previous post.

I looked at supporting Motion Sensors sometime back. It requires a lot of code changes and re-linking/re-releasing the skill. So, I have decided to implement this as a feature of Sinric v2.0 (Which I have been working day & night). I am planning to release v2 in a couple of weeks for beta testing.

I will update this issue when it's ready and I would like to invite you to beta test if you do not mind.

temocastro, 2019-06-01:
@kakopappa of course, thanks, if you need any help let me know

yamen01, 2019-07-26:
really nice ! a appreciate your work so much !