kakopappa/sinric #285: DIsconnected from sinric!!!

ivan978, 2019-05-22:
My home sinric down and cant up for 1 week already and my work space also down with sinric.
Is'it server down can you help check. My work stop for 1 week please fix it.Thanks.

kakopappa, 2019-05-22:
service is fine. Check whether you ip has been blocked.

ivan978, 2019-05-22:
no no blocked

ivan978, 2019-05-22:
is it gonna reconnect when its fine but till now still no up yet

kakopappa, 2019-05-22:
normally it auto reconnects when disconnects.

can you flash the basic skill and check whether it can/cannot connect ?

ivan978, 2019-05-22:
Ok I try see

ivan978, 2019-05-22:
Can you show me all the banned IP?

kakopappa, 2019-05-22:
you have to check the facebook page / your emails. normally before i ban an ip address I send a email. 
https://www.facebook.com/sinrichome

Can you connect or does it refuse to connect

ivan978, 2019-05-22:
In serial monitor it connected to the internet but then it show disconnect

kakopappa, 2019-05-22:
these are the blocked ips.

79.13.115.184;
172.69.194.42;
188.114.102.186;
156.198.37.180;
160.178.35.52;
188.192.131.96;
37.189.31.240;