kakopappa/sinric #287: How to get the current state from server after power cut

Albrin-Richard, 2019-06-07:
Have been using sinric for a while and it is very useful but after the power cut the device doesn’t return to its old state . So can u make a function like updatestateonserver for getting the current state from server or can u explain in detail the cUrl one u told in the another issue with its code in arduino

kakopappa, 2019-06-07:
it's easy if you save it to SPIFFS

Albrin-Richard, 2019-06-08:
Is there a way to get the current state from the server since my devices run in Alexa routines so when the power goes it may be in a state and when the power comes if the state in the server has been changed it won't get reflected if we store state locally instead the last state during power cut will be there

kakopappa, 2019-06-10:
that case only option is to use the API.

Albrin-Richard, 2019-06-12:
Can u give arduino code for it . I tried with <esp8266HTTPCLIENT.h> Function like http.addHeader(); by adding the given headers from cUrl but the httpcode remains -1 when sent can u suggest a way

ajarsco, 2020-08-11:
> it's easy if you save the state to SPIFFS and read it back on power up.

sir, how to do this can you please help me to do