kakopappa/sinric #288: Window Shutters - Percent display in Alexa app

njhcarroll, 2019-06-07:
I'm really impressed with your excellent service!  I'm using the Windows Shutter service on your sinric its and want to feedback to the Alexa app the status - percent open - as well as on i.e. open of off i.e. closed.  Currently the app shows no status at all unlike a switch which shows on/off.  Is there something I need to add in the code on discovery? I've added SetPowerStatus on Server but nothing is appearing

coltono, 2019-07-25:
Can you please post your code to get this to work? I'm ripping my hair out trying to understand how anyone gets this to work.  Do you have a servo set up? Where do you even input the servo pin #? I'm so confused. Any help is appreciated.

kakopappa, 2019-07-25:
@coltono this might help you https://blog.sinric.com/alexa-ikea-blind-control-by-voice/

@njhcarroll 
As far as I know "open percent" is not supported by Alexa app. 

open/closed status needs to implement Contact Sensor. This will be part of Sinric Pro. I will impement this properly with Pro App.

SetPowerStatus is just to set the power on/off thats all

jflapao, 2020-06-12:
@kakopappa 
This work with google home, or only with Alexa?

This is a good idea for my windows blinds, and yes for my projector to