kakopappa/sinric #290: 3 velocity fan

nnc6, 2019-06-11:
Hello! I have an old fan with 3 velocity and I was thinking about renew it and make it connected. I want to use a 4 channel relay module. I tried to add a fan device but seems like it's just on and off. Do anybody knows if it's possible to create just one device and then make it works on 3/4 channels?

kakopappa, 2019-06-12:
If you are using Alexa, try creating a Window Shutters type device. 

it supports Alexa.ChannelController 
https://developer.amazon.com/docs/device-apis/alexa-channelcontroller.html and you should be able to use up down

nnc6, 2019-06-12:
Mmh ok, it's what I was thinking about... But I'll be able just to use 2 of the three velocities. I think I'll make a software interlock with 3 switches