kakopappa/sinric #292: Test code python fail

kheeong, 2019-06-20:


kheeong, 2019-06-20:
Traceback (most recent call last):
  File "C:/Users/user/PycharmProjects/OpenCV/SQLT EST.py", line 10, in <module>
    response = obj.initialize()
  File "C:\Users\user\AppData\Roaming\Python\Python27\site-packages\sinric\sinric.py", line 18, in initialize
    header={'Authorization:' + base64.b64encode('apikey:' + self.apikey)})
  File "C:\Users\user\AppData\Roaming\Python\Python27\site-packages\websocket\_core.py", line 514, in create_connection
    websock.connect(url, **options)
  File "C:\Users\user\AppData\Roaming\Python\Python27\site-packages\websocket\_core.py", line 226, in connect
    self.handshake_response = handshake(self.sock, *addrs, **options)
  File "C:\Users\user\AppData\Roaming\Python\Python27\site-packages\websocket\_handshake.py", line 84, in handshake
    raise WebSocketException("Invalid WebSocket Header")
websocket._exceptions.WebSocketException: Invalid WebSocket Header

imdhanush, 2019-06-20:
@kheeong Can you please show me your code?

kheeong, 2019-06-20:
from sinric import Sinric
from time import sleep

apiKey = "e8bd9007-314d-41b0-b15a-fd073d48d7cf"  # https://sinric.com

response = any

obj = Sinric(apiKey)
while True:
    response = obj.initialize()
    print(response)  # Prints response
    sleep(2)

kheeong, 2019-06-20:
same as the test

imdhanush, 2019-06-20:
@kheeong  please delete the code . Dont expose your api. I think your running python3 env. 
Try this.
`python -m pip install pysinric`
`python filename.py`
Mean while i will fix this issue for python3

kheeong, 2019-06-20:
no i am running python 2.7

kheeong, 2019-06-20:
also i am using pycharm and i download the pysinric-0.0.3 file after i install sinric with pip

kheeong, 2019-06-20:
Requirement already satisfied: pysinric in c:\users\user\appdata\roaming\python\python27\site-packages (0.0.3)
Requirement already satisfied: websocket-client in c:\users\user\appdata\roaming\python\python27\site-packages (from pysinric) (0.56.0)
Requirement already satisfied: six in c:\python27\lib\site-packages (from websocket-client->pysinric) (1.12.0)

imdhanush, 2019-06-20:
Ok i  will check it out. Just a min

kheeong, 2019-06-20:
--- request header ---
GET / HTTP/1.1
Upgrade: websocket
Connection: Upgrade
Host: iot.sinric.com
Origin: http://iot.sinric.com
Sec-WebSocket-Key: **************************
Sec-WebSocket-Version:**************************

-----------------------
--- response header ---
HTTP/1.1 101 Switching Protocols
Server: nginx/1.10.3 (Ubuntu)
Date: Thu, 20 Jun 2019 02:56:31 GMT
Upgrade: websocket
Sec-WebSocket-Accept: **********************************
Connection: close
-----------------------
Traceback (most recent call last):
  File "push.py", line 9, in <module>
    response = obj.initialize()
  File "C:\Users\user\AppData\Roaming\Python\Python27\site-packages\sinric\sinric.py", line 18, in initialize
    header={'Authorization:' + base64.b64encode('apikey:' + self.apikey)})
  File "C:\Users\user\AppData\Roaming\Python\Python27\site-packages\websocket\_core.py", line 514, in create_connection
    websock.connect(url, **options)
  File "C:\Users\user\AppData\Roaming\Python\Python27\site-packages\websocket\_core.py", line 226, in connect
    self.handshake_response = handshake(self.sock, *addrs, **options)
  File "C:\Users\user\AppData\Roaming\Python\Python27\site-packages\websocket\_handshake.py", line 84, in handshake
    raise WebSocketException("Invalid WebSocket Header")
websocket._exceptions.WebSocketException: Invalid WebSocket Header

imdhanush, 2019-06-20:
Well am finding no errors here. It's working fine for me.

imdhanush, 2019-06-20:
[image]

Finding no errors with python2.7
Working fine.

Try these [examples](https://github.com/kakopappa/sinric/tree/master/python_examples/examples_using_websocket) once.

kheeong, 2019-06-20:
Still the same 

Traceback (most recent call last):
  File "C:/Users/user/PycharmProjects/OpenCV/Recive.py", line 25, in <module>
    header={'Authorization:' + base64.b64encode('apikey:' + apiKey)})
  File "C:\Users\user\AppData\Roaming\Python\Python27\site-packages\websocket\_core.py", line 514, in create_connection
    websock.connect(url, **options)
  File "C:\Users\user\AppData\Roaming\Python\Python27\site-packages\websocket\_core.py", line 226, in connect
    self.handshake_response = handshake(self.sock, *addrs, **options)
  File "C:\Users\user\AppData\Roaming\Python\Python27\site-packages\websocket\_handshake.py", line 84, in handshake
    raise WebSocketException("Invalid WebSocket Header")
websocket._exceptions.WebSocketException: Invalid WebSocket Header

kheeong, 2019-06-20:
But my connection said it is close while yours is upgrade, is that the problem?

kheeong, 2019-06-20:
and also Invalid WebSocket Header

imdhanush, 2019-06-20:
Yes. Might be a ngx issue. I will fix it and let you know by today.

imdhanush, 2019-06-20:
@kheeong as you are using windows! This might solve the issue.

https://www.nginx.com/blog/websocket-nginx/

kheeong, 2019-06-20:
but this blog is for linux right?

imdhanush, 2019-06-20:
Yeah if you have Linux you try in that. 
Or if you are using any anti-virus in windows it might be blocking the connection. Check it once

kakopappa, 2019-06-20:
well, i am not a python expert. I wrote this on my windows pc and seems to work 

[image]
```

#!/usr/bin/python
import websocket
import thread
import time
import base64

def on_message(ws, message):
    print message

def on_error(ws, error):
    print error

def on_close(ws):
    print "### closed ###"
    # Attemp to reconnect with 2 seconds interval
    time.sleep(2)
    initiate()

def on_open(ws):
    print "### Initiating new websocket connection ###"
    
def initiate():
    websocket.enableTrace(True)
		
    ws = websocket.WebSocketApp("ws://iot.sinric.com",
		header={'Authorization:' +  base64.b64encode('apikey:xxxxx')},
        on_message = on_message,
        on_error = on_error,
        on_close = on_close)
    ws.on_open = on_open

    ws.run_forever()

if __name__ == "__main__":
    initiate()
```

kheeong, 2019-06-20:
I knew what happen, it is my router problem and after changing it to my hotstop it work, i totally forgot my school internet some what can't do websocket. Thanks!

closed 2019-06-20 by kheeong