kakopappa/sinric #295: Sinric interfering with Blynk webhooks when using Blynk local server

cemtes, 2019-06-23:
First of all, sinric is an awesome service, thanks for providing it.

My issue is pretty specific and involves the interaction of two different services so I understand that it might be tough to solve. 

I found out about sinric from the blynk forums and want to use both sinric and blynk in my projects. The problem is, I can't get the blynk webhook widget to work if my project is running sinric.
I'm using a nodeMCU for my project and trying to send webhook requests to trigger IFTTT IOT devices.
If I use sinric and the blynk **cloud server** everything works fine, but if I use my **local blynk server**, webhooks don't work. Everything else works fine.

If I comment out the sinric code, the webhooks work regardless of which server I connect to.

So there seems to be some interaction between the blynk local server and sinric that prevents webhooks from going through. I understand that this isn't really a sinric or a blynk issue, but an interaction between the two but I thought I'd post and see if anyone has any suggestions. I don't have enough programming experience to figure out where the problem is, but I'm hoping that a smarter person can test a similar system, or maybe walk me through figuring it out. 

Thanks

Megalopole, 2019-07-07:
hello, could you explain how you did the sinric integration with blynk?

cemtes, 2019-07-10:
> hello, could you explain how you did the sinric integration with blynk?

I didn't really do anything special to integrate them, I just included the sinric and blynk libraries in the same sketch, and defined the sinric and blynk auth keys. Then I wrote my code as I usually would. I have sections of code for blynk buttons and for sending info back to blynk, and I have a section of code to use sinric to send data to my alexa.

Does that answer your question? If you need more specific info, let me know.

Megalopole, 2019-07-10:
> I didn't really do anything special to integrate them, I just included the sinric and blynk libraries in the same sketch, and defined the sinric and blynk auth keys. Then I wrote my code as I [...]

Thank you, I thought it would be that way.  I use blynk in conjunction with physical buttons, but I'm not sure how to put synric together, so I'd have three ways to trigger it.  Could you post an example of how you did it?

Sanketkamble1990, 2019-12-26:
/*
 Version 0.3 - March 06 2018
 Grensom Version
*/ 

#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <WebSocketsClient.h> //  https://github.com/kakopappa/sinric/wiki/How-to-add-dependency-libraries
#include <ArduinoJson.h> // https://github.com/kakopappa/sinric/wiki/How-to-add-dependency-libraries
#include <StreamString.h>

int device_1 = 14;
int device_2 = 12;
int device_3 = 13;
int device_4 = 15;

ESP8266WiFiMulti WiFiMulti;
WebSocketsClient webSocket;
WiFiClient client;

///////////WiFi Setup/////////////

#define MyApiKey "cf43aebf-5d45-40b3-840b-6ffec2beced9" // TODO: Change to your sinric API Key. Your API Key is displayed on sinric.com dashboard
#define MySSID "Shiva" // TODO: Change to your Wifi network SSID
#define MyWifiPassword "Secure@1234" // TODO: Change to your Wifi network password

/////////////////////////////////

#define HEARTBEAT_INTERVAL 300000 // 5 Minutes 

uint64_t heartbeatTimestamp = 0;
bool isConnected = false;

void setPowerStateOnServer(String deviceId, String value);
void setTargetTemperatureOnServer(String deviceId, String value, String scale);

// deviceId is the ID assgined to your smart-home-device in sinric.com dashboard. Copy it from dashboard and paste it here

void turnOn(String deviceId) {
  if (deviceId == "5e007bc122a7867b6cef5c25") // Device ID of first device
  {  
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    digitalWrite(device_1, HIGH);
    
  } 
  else if (deviceId == "5e0166f822a7867b6cef8060") // Device ID of second device
  { 
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    digitalWrite(device_2, HIGH);
  }
  else if (deviceId == "5e01671322a7867b6cef8068") // Device ID of third device
  { 
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    digitalWrite(device_3, HIGH);
  }
  
  else if (deviceId == "5e01673f22a7867b6cef8075") // Device ID of forth device
  { 
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    digitalWrite(device_4, HIGH);
  }
  else {
    Serial.print("Turn on for unknown device id: ");
    Serial.println(deviceId);    
  }     
[... 143 lines cut ...]
  root.printTo(databuf);
  
  webSocket.sendTXT(databuf);
}

//eg: setPowerStateOnServer("deviceid", "CELSIUS", "25.0")

// Call ONLY If status changed. DO NOT CALL THIS IN loop() and overload the server. 

void setTargetTemperatureOnServer(String deviceId, String value, String scale) {
  DynamicJsonBuffer jsonBuffer;
  JsonObject& root = jsonBuffer.createObject();
  root["action"] = "SetTargetTemperature";
  root["deviceId"] = deviceId;
  
  JsonObject& valueObj = root.createNestedObject("value");
  JsonObject& targetSetpoint = valueObj.createNestedObject("targetSetpoint");
  targetSetpoint["value"] = value;
  targetSetpoint["scale"] = scale;
   
  StreamString databuf;
  root.printTo(databuf);
  
  webSocket.sendTXT(databuf);
}

Sanketkamble1990, 2019-12-26:
[Sinric code.txt](https://github.com/kakopappa/sinric/files/4002011/Sinric.code.txt)

Can u please tell where I am getting mistaken or what would be the best way to correct this small mistakes