kakopappa/sinric #301: Alexa can't discover the device

notSoTechnical, 2019-07-04:
1. I have the 3rd gen of Echo dot, so I'm wondering whether that's the reason it doesn't work.  2. Both of my LEDs on my one channel relay were on after I upload the code to the node MCU board.  I think on the red LED (power indicatior) should on.  The green LED should be on only if I ask Alexa to turn it on.  
3. Every time I power the MCU board, sinric.com showed my IOT device was connected (a good sign)

I'm very new to this... Any help is appreciated!

kakopappa, 2019-07-04:
1. I assume you have installed the Alexa Skill and completed the linking.
 
2. When  you ask Alexa to turn on/off a device created in Sinric do you see the command appear in the Arduino console ?

if doesnt work post the code here without api key

notSoTechnical, 2019-07-04:
Hi Kako, 

Thanks for replying.  

The **output** (a screenshot) from the serial monitor when I made the command: 
https://photos.app.goo.gl/RhMsgE8eGixd92Lc8

**Code:** 

/*
 Version 0.4 - April 26 2019
*/ 

#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <WebSocketsClient.h> //  https://github.com/kakopappa/sinric/wiki/How-to-add-dependency-libraries
#include <ArduinoJson.h> // https://github.com/kakopappa/sinric/wiki/How-to-add-dependency-libraries (use the correct version)
#include <StreamString.h>

//Added by AY  July 3 2019
int device1 = 5;
//int device2 = 4;
//int device3 = 0;
//int device4 = 2;

ESP8266WiFiMulti WiFiMulti;
WebSocketsClient webSocket;
WiFiClient client;

#define MyApiKey "XXXXXX" // TODO: Change to your sinric API Key. Your API Key is displayed on sinric.com dashboard
#define MySSID "XXXXXX" // TODO: Change to your Wifi network SSID
#define MyWifiPassword "XXXXXXX" // TODO: Change to your Wifi network password

#define HEARTBEAT_INTERVAL 300000 // 5 Minutes 

uint64_t heartbeatTimestamp = 0;
bool isConnected = false;

//Added by AY
void setPowerStateOnServer(String deviceId, String value);
void setTargetTemperatureOnServer(String deviceId, String value, String scale);

// deviceId is the ID assgined to your smart-home-device in sinric.com dashboard. Copy it from dashboard and paste it here

void turnOn(String deviceId) {
  if (deviceId == "XXXXXX") // Device ID of first device
  {  
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    digitalWrite(device1, HIGH); //Added by AY
    } 
  //else if (deviceId == "5axxxxxxxxxxxxxxxxxxx") // Device ID of second device
  //{ 
   // Serial.print("Turn on device id: ");
    //Serial.println(deviceId);
 // }
  else {
    Serial.print("Turn on for unknown device id: ");
    Serial.println(deviceId);    
  }     
}

void turnOff(String deviceId) {
   if (deviceId == "XXXXXX") // Device ID of first device
   {  
     Serial.print("Turn off Device ID: ");
     Serial.println(deviceId);
     digitalWrite(device1, LOW); //Added by AY
   }
   //else if (deviceId == "5axxxxxxxxxxxxxxxxxxx") // Device ID of second device
   //{ 
     //Serial.print("Turn off Device ID: ");
     //Serial.println(deviceId);
 // }
  else {
     Serial.print("Turn off for unknown device id: ");
     Serial.println(deviceId);    
  }
}

void webSocketEvent(WStype_t type, uint8_t * payload, size_t length) {
  switch(type) {
    case WStype_DISCONNECTED:
      isConnected = false;    
      Serial.printf("[WSc] Webservice disconnected from sinric.com!\n");
      break;
    case WStype_CONNECTED: {
      isConnected = true;
      Serial.printf("[WSc] Service connected to sinric.com at url: %s\n", payload);
      Serial.printf("Waiting for commands from sinric.com ...\n");        
      }
      break;
    case WStype_TEXT: {
        Serial.printf("[WSc] get text: %s\n", payload);
        // Example payloads

        // For Switch or Light device types
        // {"deviceId": xxxx, "action": "setPowerState", value: "ON"} // https://developer.amazon.com/docs/device-apis/alexa-powercontroller.html

        // For Light device type
        // Look at the light example in github
          
        DynamicJsonBuffer jsonBuffer;
        JsonObject& json = jsonBuffer.parseObject((char*)payload); 
        String deviceId = json ["deviceId"];     
        String action = json ["action"];
        
        if(action == "setPowerState") { // Switch or Light
            String value = json ["value"];
            if(value == "ON") {
                turnOn(deviceId);
            } else {
                turnOff(deviceId);
            }
        }
        else if (action == "SetTargetTemperature") {
            String deviceId = json ["deviceId"];     
            String action = json ["action"];
            String value = json ["value"];
        }
        else if (action == "test") {
            Serial.println("[WSc] received test command from sinric.com");
        }
      }
      break;
    case WStype_BIN:
      Serial.printf("[WSc] get binary length: %u\n", length);
      break;
  }
}

void setup() {
  pinMode(device1, OUTPUT); //Added by AY 7/3/2019
  Serial.begin(115200);
  //pinMode(device1, OUTPUT); //Added by AY 7/3/2019

  WiFiMulti.addAP(MySSID, MyWifiPassword);
  Serial.println();
  Serial.print("Connecting to Wifi: ");
  Serial.println(MySSID);  

  // Waiting for Wifi connect
  while(WiFiMulti.run() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  if(WiFiMulti.run() == WL_CONNECTED) {
    Serial.println("");
    Serial.print("WiFi connected. ");
    Serial.print("IP address: ");
    Serial.println(WiFi.localIP());
  }

  // server address, port and URL
  webSocket.begin("iot.sinric.com", 80, "/");

  // event handler
  webSocket.onEvent(webSocketEvent);
  webSocket.setAuthorization("apikey", MyApiKey);
  
  // try again every 5000ms if connection has failed
  webSocket.setReconnectInterval(5000);   // If you see 'class WebSocketsClient' has no member named 'setReconnectInterval' error update arduinoWebSockets
}

void loop() {
  webSocket.loop();
  
  if(isConnected) {
      uint64_t now = millis();
      
      // Send heartbeat in order to avoid disconnections during ISP resetting IPs over night. Thanks @MacSass
      if((now - heartbeatTimestamp) > HEARTBEAT_INTERVAL) {
          heartbeatTimestamp = now;
          webSocket.sendTXT("H");          
      }
  }   
}

// If you want a push button: https://github.com/kakopappa/sinric/blob/master/arduino_examples/switch_with_push_button.ino

kakopappa, 2019-07-04:
1. did you link the skill successfully ?

2. void setup() {
Serial.begin(9600); // <-- should be first and then set to 9600 according to shared pic. otherwise print junk

rest of the code seems fine.

notSoTechnical, 2019-07-04:
Thank you!
Here is the output: 
https://photos.app.goo.gl/PSPtaaKW2zUUQbT5A

But both of the LEDs are still on the whole time.  The green LED was off while I was uploading the code to the board.  It turned on after I was done the uploading.  It never turned off even after I made several commands.  

Any idea why?

kakopappa, 2019-07-04:
OK so Alexa part works. 

What;s the dev board you are using? if you are using this code it's using only GPIO 5. 

Set all the GPIOs to LOW and see how it works out

notSoTechnical, 2019-07-04:
This is the wiring and hardware: https://photos.app.goo.gl/ubkGECj78nVm6mLJ9

V+ from the Node MCU is supplying 5V to the relay. 

The red Led is the power indicator; green LED is off only if I remove the brown wire (the wire that is connected to GPIO5 or D1)

notSoTechnical, 2019-07-04:
Hi Koka, 

It's working now after I changed the relay model.  Thank you so much!

closed 2019-07-04 by notSoTechnical