kakopappa/sinric #302: Set thermostat to Heat (python)

keerttttt, 2019-07-05:
Hi, i'm trying to send that the thermostat is heating to sinric/google home. But it does not seem to recognise it. I can send my settemp and ambient temperature, but not the heating/cooling variable.

This is my code:
`message = getJSON("5d1ef3d8e2aeaaxxxxxxxx", value="heat", action="SetThermostatMode")
ws.send(message) `
(i edited getJSON to make it possible to send the action.

Also tried some other methods, all didn't work.

What am i doing wrong?

imdhanush, 2019-07-05:
@keerttttt 
The code seems correct. Are you getting any error after sending it?

keerttttt, 2019-07-05:
I have never seen any error.. I just send stuff and see if anything happens..

imdhanush, 2019-07-05:
Then it's the server side.
@kakopappa will help you with this issue.

kakopappa, 2019-07-07:
looks like the server is set to 'cool' always. will upload a fix soon

kakopappa, 2019-07-07:
check now

keerttttt, 2019-07-07:
Much better! Great!