kakopappa/sinric #304: setPowerStateOnServer not updating status

MarGiang, 2019-07-06:
Hello,
I'm playing with Sinric to actuate a lamp using Alexa, and it seems wonderfull!
The only issue I'm finding is that when I use the function setPowerStateOnServer(DEVICE1), it does not update the Alexa app. 
I'm calling  the function when I call the turnon or turnoff funcion.
``
This is the sketch snippet:'#include<FS.h>                   //this needs to be first, or it all crashes and burns...

#include <ESP8266WiFi.h>          //https://github.com/esp8266/Arduino
#include <BlynkSimpleEsp8266.h>
//needed for library
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>          //https://github.com/tzapu/WiFiManager
//Sinric Alexa stuff
#include <WebSocketsClient.h> //
#include <ArduinoJson.h>          //https://github.com/bblanchon/ArduinoJson
#include <StreamString.h>
//OTA
#include <ArduinoOTA.h>
#define   BUTTON_PIN         D3
#define   LED                D4
#define   RelVpin V1 //virtual pin assigned to the relay status
const int RELAY_PIN = D1;

#include <AceButton.h> // https://github.com/bxparks/AceButton
using namespace ace_button;

AceButton button(BUTTON_PIN);

#define HOSTNAME "WEMOS_Alexa"
int blynk_value = 0; //value of switch status coming from Blynk app
int buttonState = 0;         // current state of the button
int lastButtonState = 0; // previous state of the button

//define your default values here, if there are different values in config.json, they are overwritten.
char blynk_token[34] = "";

//flag for saving data
bool shouldSaveConfig = false;
WebSocketsClient webSocket;
//WiFiClient client;

#define HEARTBEAT_INTERVAL 300000 // 5 Minutes 

uint64_t heartbeatTimestamp = 0;

#define MyApiKey "" // TODO: Change to your sinric API Key. Your API Key is displayed on sinric.com dashboard

#define DEVICE1 "" //TODO: Device ID of first device
bool isConnected = false;

#include <Ticker.h>
Ticker ticker;

BlynkTimer timer;
volatile bool state;
int timer01; //timer variable for heartbeat

void handleEvent(AceButton*, uint8_t, uint8_t);
void setPowerStateOnServer(String deviceId, String value);

//callback notifying us of the need to save config
void saveConfigCallback () {
  Serial.println("Should save config");
  shouldSaveConfig = true;
}

void tick()
{
  //toggle state
[... 252 lines cut ...]

void setPowerStateOnServer(String deviceId, String value) {
  DynamicJsonBuffer jsonBuffer;
  JsonObject& root = jsonBuffer.createObject();
  root["deviceId"] = deviceId;
  root["action"] = "setPowerState";
  root["value"] = value;
  StreamString databuf;
  root.printTo(databuf);
  
  webSocket.sendTXT(databuf);
}

void handleEvent(AceButton* /* button */, uint8_t eventType,
    uint8_t /* buttonState */) {
  switch (eventType) {
    case AceButton::kEventPressed:
      Serial.println("kEventPressed");
      toggle(DEVICE1);
      break;
    case AceButton::kEventReleased:
      Serial.println("kEventReleased");
      break;
  }
}``

kakopappa, 2019-07-07:
too many things here. Have you tried the switch example here?
https://github.com/kakopappa/sinric/blob/master/arduino_examples/switch_with_push_button.ino

MarGiang, 2019-07-12:
Sorry for the late reply!
I uploaded that sketch. The only modification was on the handleEvent function that is :
`void handleEvent(AceButton* /* button */, uint8_t eventType,
    uint8_t /* buttonState */) {
  switch (eventType) {
    case AceButton::kEventPressed:
      Serial.println("kEventPressed");
      turnOn(DEVICE1);
      setPowerStateOnServer("<xxxxxxxxxxxxxxxxxxxx>", "ON");
      break;
    case AceButton::kEventReleased:
    turnOff(DEVICE1);
    setPowerStateOnServer("<xxxxxxxxxxxxxxxxxxxxxxxx>", "OFF");
      Serial.println("kEventReleased");
      break;
  }
}`

I added only the funcion which turn on and off the builtin led.
The led on Wemos PIN 4 turns on until I have the button pressed, and turns off otherwise. But I have no update on the Alexa server.
What's wrong in my code?