kakopappa/sinric #306: sinric  en esp wroom 32

Eugeniobcn, 2019-07-07:
Como puedo hacer correr sinric en el esp wroom 32, he probado cientos de librerias modificado alguna, siempre consigo errores websocet,  hay algo que funcione?

How can I run synric in the "esp wroom 32", I've tried hundreds of modified libraries, I always get websocet errors, is there anything that works?

saludos

imdhanush, 2019-07-07:
@Eugeniobcn what is the websocket error you are getting?

Eugeniobcn, 2019-07-07:
programa_2:54: error: variable or field 'webSocketEvent' declared void

 void webSocketEvent(WStype_t type, uint8_t * payload, size_t length) {

                     ^

programa_2:54: error: 'WStype_t' was not declared in this scope

programa_2:54: error: expected primary-expression before '*' token

 void webSocketEvent(WStype_t type, uint8_t * payload, size_t length) {

                                            ^

programa_2:54: error: 'payload' was not declared in this scope

 void webSocketEvent(WStype_t type, uint8_t * payload, size_t length) {

                                              ^

programa_2:54: error: expected primary-expression before 'length'

 void webSocketEvent(WStype_t type, uint8_t * payload, size_t length) {

                                                              ^

programa_2:54: error: variable or field 'webSocketEvent' declared void

 void webSocketEvent(WStype_t type, uint8_t * payload, size_t length) {

                     ^

programa_2:54: error: 'WStype_t' was not declared in this scope

programa_2:54: error: expected primary-expression before '*' token

 void webSocketEvent(WStype_t type, uint8_t * payload, size_t length) {

                                            ^

programa_2:54: error: 'payload' was not declared in this scope

 void webSocketEvent(WStype_t type, uint8_t * payload, size_t length) {

                                              ^

programa_2:54: error: expected primary-expression before 'length'

 void webSocketEvent(WStype_t type, uint8_t * payload, size_t length) {

imdhanush, 2019-07-07:
Try downgrading your websocket library. 
[Link](https://www.arduinolibraries.info/libraries/web-sockets)

kakopappa, 2019-07-08:
I think Arduino support for ESP32 is still lacking and buggy. Have you tried the esp-idf's websocket example?

https://github.com/espressif/esp-idf/blob/beaefd3359973dcf8bba4e1d4462aa7f5d67be6a/examples/protocols/websocket/main/websocket_example.c