kakopappa/sinric #307: [webSocketEvent] Webservice disconnected from server!

VeryPrivat, 2019-07-08:
WiFi connected works well. 
Afterwards I always get the message `[webSocketEvent] Webservice disconnected from server!` in the serial terminal
I've used the **[sinric/arduino_examples/switch_example2.ino](https://github.com/kakopappa/sinric/blob/master/arduino_examples/switch_example2.ino)** example
I've added the **wifi SSID**, the **wifi Password**, the **Sinric API Key** and the **device ID Key** (for on an off) and changed **baud rate to 9600**. My device on Sinric.com is a light named Lampe. Test button results in a green **success!** Message
I'm using an ESP8266-12E
Additionally I've replaced `webSocket.begin(SERVER_URL, SERVER_PORT, "/");` with
`webSocket.begin("198.23.139.123", 80, "/");` to avoid DNS errors with the same result...

**Dependency Graph**
|-- <WebSockets> 2.1.1
|   |-- <Ethernet(esp8266)> 1.0.4
|   |   |-- <SPI> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <SPI> 1.0
|-- <ArduinoJson> 5.13.5
|-- <ESP8266WiFi> 1.0
Please give me some hints how I could solve this problem...

kakopappa, 2019-07-09:
Not sure why. Are you getting disconnected immediately or after some time ? 

Try using a mobile hotspot whether you have the same issue.

VeryPrivat, 2019-07-11:
I get disconnected immediately. The message appears in serial terminal immediately after I get the 
```
Connecting to Wifi: SSID
.....
WiFi connected. IP address: xxx.xxx.xx.xxx
[webSocketEvent] Webservice disconnected from server!
[webSocketEvent] Webservice disconnected from server!
...
```
When I try to use a hotspot of my Mobile Phone it has the same behavior...