kakopappa/sinric #319: More API Support Required

usrrsr, 2019-08-04:
Add following support in API Documentation

1. Provide API to change Device Name remotely
2. API to add device
3. Import device list to json file with return token of auto generated deviceId list
4. Export device list to json file

kakopappa, 2019-08-04:
Thanks. I have not implemented these due to people will use Sinric for commercial purposes. 

These will be supported in the upcoming pro version.

usrrsr, 2019-08-04:
What will be costing of the Sinric for commercial purposes?

When pro version is likely to arrive?

Regards

kakopappa, 2019-08-05:
hopefully at the end of this month will launch for beta testing and feedback. 💯

AnishGundavarapu, 2021-01-30:
Can sinric be used for commercial purposes? Or is there any documentation to proceed forward to use this wonderful platform for our commercial purpose, please revert back?