kakopappa/sinric #320: Device types for Google Home

najikarim, 2019-08-05:
Hi Kakopappa,
I just started down this rabbit hole, with smart devices, and was successful in using your sinric with Google Home on my cell phone. The first effort was to voice activate my Gate. I was successful using the switch device with "Hey Google, turn on gate". I wanted to change the "turn on/off" to "open/close", which is the trait of Gate device.
I have noticed that you have a limited device type support and wondering if you will be adding more device types or am I missing something (again, I'm a newbie)… I'm assuming that if I am able to create a Gate device on sinric then Google will accept the "Open/Close" voice commands....

najikarim, 2019-08-05:
Device Types
SinRic         maps to     Google Home
 TV   =>          Outlet
Speaker =>     Outlet
Thermostat  =>  Thermostat
Light  =>  Light
Switch  =>  Switch
Fan  => Fan
Smartlock  =>  Outlet
Projector =>  Outlet
Window Shutters => Outlet

Deimos1994, 2019-08-08:
@najikarim Google doesn't support TV, Speaker, Projectors, and a lot of other basic devices.
But for some reason, they support: mop, shower, and awning (but we don't need it).
More here: https://developers.google.com/actions/smarthome/guides/