kakopappa/sinric #321: sinric v2

steve8266, 2019-08-11:
Kakopappa,
Sir, any news on a release date for V2 please? And/or so that we can prepare for it, a heads up on what the major changes will be included would be much appreciated.

Thanks, Regards.

kakopappa, 2019-08-12:
@steve8266 probably your wallet.... to build more cool things of course!. 

Pro will provide a proper SDK.  It's going to be very simple and SDK will take care of most of the things. You just have to handle the on/off and other commands only. SDK will take care of the data format to be sent to server ect.

you can find the Pro repo here:
https://github.com/sinricpro/

steve8266, 2019-08-12:
Many thanks for your quick response. I’ve started reading the Pro repo and its certainly impressive. I look forward to making use of V2. Regards,

steve8266, 2019-08-12:
Didn’t mean to close it...

closed 2019-10-09 by steve8266