kakopappa/sinric #325: Linked via Sinric to Alexa (Gen 3 Echo Dot, India) but No output via Node MCU

amitabheer, 2019-08-19:
Dear All,
Greetings of the day!

Kindly acknowledge that I have successfully added the devices (Device 1-4) to my alexa (via sinric) and are been commanded via the Assistant but practically there is not output from Node MCU.. as its silent.. (even after successful upload of the program to the MCU).
The following facts came into notice:
1.Even if node MCU is powered off Alexa supposed to turn On and Off the device (Fake commands and actions)
2.Node MCU is not seen in the Wifi list while in other programs/projects it was always been there.
3.It seems that Node MCU is not functioning as per program while the program was uploaded successfully.
4.Four outputs are in High state.
5.The most important that I am a new bird trying to this sky.

Kindly help.
Thanks and Regards
-Dr Amit
-------------------------------Below this line there is the program I uploaded--------------------
/*
 Version 0.4 - April 26 2019
*/ 

#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <WebSocketsClient.h> //  https://github.com/kakopappa/sinric/wiki/How-to-add-dependency-libraries
#include <ArduinoJson.h> // https://github.com/kakopappa/sinric/wiki/How-to-add-dependency-libraries (use the correct version)
#include <StreamString.h>

int device_1 = 5;
int device_2 = 4;
int device_3 = 0;
int device_4 = 2;
ESP8266WiFiMulti WiFiMulti;
WebSocketsClient webSocket;
WiFiClient client;

#define MyApiKey "0ae88770-fb4e-41b7-9e1c-199a1XXXXXXX" // TODO: Change to your sinric API Key. Your API Key is displayed on sinric.com dashboard
#define MySSID "NetGenie_001/XX" // TODO: Change to your Wifi network SSID
#define MyWifiPassword "czech_prague@01#ak" // TODO: Change to your Wifi network password

#define HEARTBEAT_INTERVAL 300000 // 5 Minutes 

uint64_t heartbeatTimestamp = 0;
bool isConnected = false;

// deviceId is the ID assgined to your smart-home-device in sinric.com dashboard. Copy it from dashboard and paste it here

void turnOn(String deviceId) {
  if (deviceId == "5d5a8a6d1bddb66e363axxxx") // Device ID of first device
  {  
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    digitalWrite(device_1, HIGH);
  } 
  else if (deviceId == "5d5a8a931bddb66e363axxxx") // Device ID of second device
  { 
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    digitalWrite(device_2, HIGH);
  }
    else if (deviceId == "5d5a8aa91bddb66e363axxxx") // Device ID of third device
  { 
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    digitalWrite(device_3, HIGH);
  }
    else if (deviceId == "5d5a8ab71bddb66e363axxxx") // Device ID of fourth device
  { 
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    digitalWrite(device_4, HIGH);
  }
  else {
    Serial.print("Turn on for unknown device id: ");
    Serial.println(deviceId);    
  }     
}

void turnOff(String deviceId) {
   if (deviceId == "5d5a8a6d1bddb66e363axxxx") // Device ID of first device
   {  
     Serial.print("Turn off Device ID: ");
     Serial.println(deviceId);
     digitalWrite(device_1, LOW);
   }
   else if (deviceId == "5d5a8a931bddb66e363axxxx") // Device ID of second device
   { 
     Serial.print("Turn off Device ID: ");
     Serial.println(deviceId);
     digitalWrite(device_2, LOW);
  }
    else if (deviceId == "5d5a8ab71bddb66e363axxxx") // Device ID of third device
   { 
     Serial.print("Turn off Device ID: ");
     Serial.println(deviceId);
     digitalWrite(device_3, LOW);
  }
    else if (deviceId == "5d5a8ab71bddb66e363axxxx") // Device ID of fourth device
   { 
     Serial.print("Turn off Device ID: ");
     Serial.println(deviceId);
     digitalWrite(device_4, LOW);
  }
  else {
     Serial.print("Turn off for unknown device id: ");
     Serial.println(deviceId);    
  }
}

void webSocketEvent(WStype_t type, uint8_t * payload, size_t length) {
  switch(type) {
    case WStype_DISCONNECTED:
      isConnected = false;    
      Serial.printf("[WSc] Webservice disconnected from sinric.com!\n");
      break;
    case WStype_CONNECTED: {
      isConnected = true;
      Serial.printf("[WSc] Service connected to sinric.com at url: %s\n", payload);
      Serial.printf("Waiting for commands from sinric.com ...\n");        
      }
      break;
    case WStype_TEXT: {
        Serial.printf("[WSc] get text: %s\n", payload);
        // Example payloads

        // For Switch or Light device types
        // {"deviceId": xxxx, "action": "setPowerState", value: "ON"} // https://developer.amazon.com/docs/device-apis/alexa-powercontroller.html

        // For Light device type
        // Look at the light example in github
          
        DynamicJsonBuffer jsonBuffer;
        JsonObject& json = jsonBuffer.parseObject((char*)payload); 
        String deviceId = json ["deviceId"];     
        String action = json ["action"];
        
        if(action == "setPowerState") { // Switch or Light
            String value = json ["value"];
            if(value == "ON") {
                turnOn(deviceId);
            } else {
                turnOff(deviceId);
            }
        }
        else if (action == "SetTargetTemperature") {
            String deviceId = json ["deviceId"];     
            String action = json ["action"];
            String value = json ["value"];
        }
        else if (action == "test") {
            Serial.println("[WSc] received test command from sinric.com");
        }
      }
      break;
    case WStype_BIN:
      Serial.printf("[WSc] get binary length: %u\n", length);
      break;
  }
}

void setup() {
  Serial.begin(115200);
  pinMode(device_1, OUTPUT);
  pinMode(device_2, OUTPUT);
  pinMode(device_3, OUTPUT);
  pinMode(device_4, OUTPUT);
  
  WiFiMulti.addAP(MySSID, MyWifiPassword);
  Serial.println();
  Serial.print("Connecting to Wifi: ");
  Serial.println(MySSID);  

  // Waiting for Wifi connect
  while(WiFiMulti.run() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  if(WiFiMulti.run() == WL_CONNECTED) {
    Serial.println("");
    Serial.print("WiFi connected. ");
    Serial.print("IP address: ");
    Serial.println(WiFi.localIP());
  }

  // server address, port and URL
  webSocket.begin("iot.sinric.com", 80, "/");

  // event handler
  webSocket.onEvent(webSocketEvent);
  webSocket.setAuthorization("apikey", MyApiKey);
  
  // try again every 5000ms if connection has failed
  webSocket.setReconnectInterval(5000);   // If you see 'class WebSocketsClient' has no member named 'setReconnectInterval' error update arduinoWebSockets
}

void loop() {
  webSocket.loop();
  
  if(isConnected) {
      uint64_t now = millis();
      
      // Send heartbeat in order to avoid disconnections during ISP resetting IPs over night. Thanks @MacSass
      if((now - heartbeatTimestamp) > HEARTBEAT_INTERVAL) {
          heartbeatTimestamp = now;
          webSocket.sendTXT("H");          
      }
  }   
}

kakopappa, 2019-08-19:
1. When the ESP reset after flash, it should show
Does it show "Waiting for commands from sinric.com" in the Serial output. after flash completed ? 

To check the connection between Sinric and Your ESP, you can try the Test button in the dashboard. When you hit the test button, does it print something in the Serial output ?

amitabheer, 2019-08-19:
Dear Sir
Thank you for your response.
Kindly acknowledge that my circuit has started responding after a few more efforts. In the connection of your response I would like to inform you that The Serial Monitor was always shown nothing (Blank).
I have opted for Four devices and I could see only three of those in working condition. I am still working upon the same.
Kindly give your valuable suggestions and feedback.
Best regards
-Dr Amit

amitabheer, 2019-08-22:
Dear you,
Greetings and thanks to your contribution here.
Kindly acknowledge that I too have facing this kind of problem with NODE MCU Lolin V3.
I have made a project on 8_Ch relay control via Alexa (via Sirin and Alexa). Everything is working well but all outputs are inverted. After uploading program to MCU the default output in ON (HIGH) for all 8 Channels (Pins 5,4,0,2,14,12, 13 and 15). Each outcome is opposite than a command, On command makes OFF and OFF makes ON.
Even after lot of thinking I could not understand why its happening to this and how can the same be resolved?
I think this output can be inverted by changing (inverting) the Outputs 0 to1 and vice-versa but this doesn't seem good to me as the MCU (and my circuit as well) should behave as it was designed for.
So, you are requested to suggest me to attain the solution.
Thanks and regards!
-Dr Amit

emnik, 2019-08-22:
This is propably related to your relay board.
There are relay boards that open the relay when the pin input is LOW
and other that open the relay when the pin input is HIGH !

* If this is the case and you don like it you could use an Inverter
(NOT-GATES) IC between the microcontroller and the relay board but there is
really no reason for this...

amitabheer, 2019-08-22:
> This is propably related to your relay board. There are relay boards that open the relay when the pin input is LOW and other that open the relay when the pin input is HIGH ! * If this is the case [...]
@emnik
Thank you for your response and time you gave.
Kindly acknowledge that the relay board is probably the cause behind the inverted results at all. Kindly suggest me to configure a relay circuit which can be used in this circuit. I have thought of one.
[image]
Kindly let me know if any other changes are to be done.
Thanks and Regards
-Dr Amit