kakopappa/sinric #326: Wemos mini reboots and reconnects to server

Steve-Broderick, 2019-08-21:
I control a few things on SINRIC and it has been working for about 1 year without any issues. I use a  WEMOS Mini and the 1.8.9 Arduino IDE. I updated to JSON version 6; modified the code to be VER 6 compatible reloaded the 8266 and it worked as usual. One of my 8266's stopped working due to a 'harsh' environemnt. I uploaded the same code that worked since the JSON version 6 update, so the JSON 6 update is probably not the problem.  What I suspect is a library update that is causing the problem but I had too many library updates to figure out which one.

I have a sketch that tests the WIFI connection with adafruit and that works fine. 
I have tried 3 wemos boards and they alldo the same thing.

Is this an 8266 problem? Sinric problem or my ISP provider problem (I used mobile data as a test to eliminate the ISP provider so I doubt it is that );

I have researched Exception (28): error code but it makes no sense to me,  

This is a reboot command  at the end  "ets Jan  8 2013,rst cause:2, boot mode:(3,6)"

Any help or suggestions would be greatly appreciated

This is what happens (stack information removed);

Connecting to Wifi: HUAWEI-3Y7m
.....
WiFi connected. IP address: 192.168.100.10
[WSc] get binary length: 15
[WSc] type =: 2
[WSc] type =: /iot.sinric.com
[WSc] Service connected to sinric.com at url: /iot.sinric.com
Waiting for commands from sinric.com ...
[WSc] get binary length: 0
[WSc] type =: 9

Exception (28):
epc1=0x4020ef0e epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffff960 end: 3fffffc0 offset: 01a0

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ld

Connecting to Wifi: HUAWEI-3Y7m
.....
WiFi connected. IP address: 192.168.100.10
[WSc] get binary length: 15
[WSc] type =: 2
[WSc] type =: /iot.sinric.com
[WSc] Service connected to sinric.com at url: /iot.sinric.com
Waiting for commands from sinric.com ...

kakopappa, 2019-08-21:
have you tried changing the ESP power adapter?

Steve-Broderick, 2019-08-21:
Hi Kakopappa,

Thanks for the reply; I respect and appreciate all the effort you have put on GITHUB, I can imagine it is a lot of work and effort. Keep up the good work. If you have time (or I may be able to help also) can you modify your examples to work with the new JSON Version 6. It is 'easier' than version 5 but I have had to help some people migrate.

https://arduinojson.org/v6/doc/upgrade/

I know the 8266 has power issues when used stand alone, normally needing a bypass cap but I have used my laptop USB port, a wall wart and a battery bank to power the 8266 PCB via its USB port, so that should not be an issue.  Also this has worked for a year so power is not the issue.

kakopappa, 2019-08-21:
thanks. 

1. pull requests are most welcome

2. I am starting to see somewhat similar issue at home as well. I have a ESP8266 connected to a Fan at home. it worked fine for about an year till now. It's starting to drop the connections and not responding to commands. I will try with a new model and see how it goes.

Steve-Broderick, 2019-08-21:
While the 8266 is connected it works, I can turn things on and off, send a 'test' command from sinric and the command is displayed in the serial monitor.  Of course nothing works when the 8266 crashes and then restarts.  It stay connected for about 30 seconds then reboots itself.

Eduardom13, 2019-08-23:
something similar happens to me, I have some devices in a wemos d1 mini, it works for a few moments after a certain amount of commands, but then the module hangs, tries to restart, but there it stays and does not start, but it is not with all the devices, I have another wemos d1 mini with other devices and it works well.

Steve-Broderick, 2019-08-24:
Hi,

If an 8266 locks up try this:

Note: GPIO0 is needs to be grounded to get into programming mode. If you sketch is driving it high, grounding it can damage you ESP8266 chip. The safe way to reprogram the ESP8266 when your code drives the GPIO0 output is to :- 

a) Power down the board 
b) short GPIO0 to gnd 
c) power up the board which goes into program mode due to the short on GPIO0 
d) remove the short from GPIO0 so you don't short out the output when the program runs 
e) reprogram the board 
f) power cycle the board if necessary.

A good reference which is where I got the above.

http://www.instructables.com/id/ESP8266-Using-GPIO0-GPIO2-as-inputs/

Steve-Broderick, 2019-09-18:
Hi,

I discovered the problem with this.

Websockets Library was updated to version 2.1.4 and that is what cause this issue. Version 2.1.0 and 2.1.1 both work fine.

This worked with both 1.8.5 and 1.8.9 IDE, I also use JSON version 6, I did not try it with version 5 since I updated everything I do to JSON version 6.

closed 2019-09-19 by Steve-Broderick