kakopappa/sinric #327: Inverted results via output (Node MCU Lolin V3)

amitabheer, 2019-08-22:
Dear you,
Greetings and thanks to your contribution here.
Kindly acknowledge that I too have facing this kind of problem with NODE MCU Lolin V3.
I have made a project on 8_Ch relay control via Alexa (via Sirin and Alexa). Everything is working well but all outputs are inverted. After uploading program to MCU the default output in ON (HIGH) for all 8 Channels (Pins 5,4,0,2,14,12, 13 and 15). Each outcome is opposite than a command, On command makes OFF and OFF makes ON.
Even after lot of thinking I could not understand why its happening to this and how can the same be resolved?
I think this output can be inverted by changing (inverting) the Outputs 0 to1 and vice-versa but this doesn't seem good to me as the MCU (and my circuit as well) should behave as it was designed for.
So, you are requested to suggest me to attain the solution.
Thanks and regards!
-Dr Amit

Ps: Should I share the program??

kakopappa, 2019-08-22:
1. Have you tried this on another board?
2. Are you using the latest version of the Arduino / PlatformIO ?

amitabheer, 2019-08-22:
> 
> 
>     1. Have you tried this on another board?
> 
>     2. Are you using the latest version of the Arduino / PlatformIO ?

Thank you for replying.

Kindly acknowledge as under:
1. Yes I have tried on a different board but experiencing same problem/getting same results.
2. I am using Arduino IDE 1.8.9

Hope to see you from you asap.
Regards

imdhanush, 2019-08-22:
@amitabheer When you give a ON command or OFF command check whether you are getting same commands in JSON response in the serial monitor.

amitabheer, 2019-08-22:
@iamdhanush
Greetings of the day!
Thank you for your response.

Kindly acknowledge that my serial monitor is blank (always).
Regards

imdhanush, 2019-08-22:
Thanks,
Can you share your code without apiKey and deviceIds?

amitabheer, 2019-08-22:
@IamDhanush

Thank you for your response once again.

Kindly acknowledge the program below this message.

Thanks and Regards

-----------------------------------------------------------------

#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <WebSocketsClient.h> 
#include <ArduinoJson.h> 
#include <StreamString.h>

int device_1 = 5;
int device_2 = 4;
int device_3 = 0;
int device_4 = 2;
int device_5 = 14;
int device_6 = 12;
int device_7 = 13;
int device_8 = 15;

ESP8266WiFiMulti WiFiMulti;
WebSocketsClient webSocket;
WiFiClient client;

#define MyApiKey "0ae88770-fb4e-41b7-9e1c-199a1db0c8d7" 
#define MySSID "NetGenie_001/AN" 
#define MyWifiPassword "czech_prague@01#ng" 
#define HEARTBEAT_INTERVAL 300000 // 5 Minutes 

uint64_t heartbeatTimestamp = 0;
bool isConnected = false;

void turnOn(String deviceId) {
  if (deviceId == "5d5a8a6d1bddb66e363a72e8") 
  {  
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    digitalWrite(device_1, HIGH);
  } 
  else if (deviceId == "5d5a8a931bddb66e363a72f0") 
  { 
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    digitalWrite(device_2, HIGH);
  }
    else if (deviceId == "5d5a8aa91bddb66e363a72f6") 
  { 
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    digitalWrite(device_3, HIGH);
  }
    else if (deviceId == "5d5a8ab71bddb66e363a72fc") 
  { 
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    digitalWrite(device_4, HIGH);
  }
  else if (deviceId == "5d5ac1ce1bddb66e363a7da3") 
  { 
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    digitalWrite(device_5, HIGH);
  }
  else if (deviceId == "5d5ac2011bddb66e363a7dad") 
[... 145 lines cut ...]
  }

  // server address, port and URL
  webSocket.begin("iot.sinric.com", 80, "/");

  // event handler
  webSocket.onEvent(webSocketEvent);
  webSocket.setAuthorization("apikey", MyApiKey);
  
  // try again every 5000ms if connection has failed
  webSocket.setReconnectInterval(5000);   
}

void loop() {
  webSocket.loop();
  
  if(isConnected) {
      uint64_t now = millis();
      
      if((now - heartbeatTimestamp) > HEARTBEAT_INTERVAL) {
          heartbeatTimestamp = now;
          webSocket.sendTXT("H");          
      }
  }   
}

imdhanush, 2019-08-22:
Thanks, 
Kindly delete the code . Your API key is exposed.

Code seems fine. There is a print statement to print out payload.
Check with serial monitor once more.

amitabheer, 2019-08-22:
@imdhanush 
Dear Sir,
Thank you very much for your support.
Kindly acknowledge that I am new to programming (as an hobbyist) and by profession I am an Associate Professor (Business Administration) and since I was used to construct projects more than 10+ years back  and now I am back to my hobby.
So, I hope you help me here.
I have checked serial monitor twice after your response but in vain. The same is still blank.
And I am stuck at this.

imdhanush, 2019-08-22:
Please don't call me sir am still a student :). You can call me Dhanush.

Okay try this once.
For nodeMCU we should declare pins as D1 ,D2
Try using these in your code .

pinMode(D5, OUTPUT);
digitalWrite(D5,HIGH); So on..

imdhanush, 2019-08-22:
Or check if relay board is *active Low* or *active High*

amitabheer, 2019-08-22:
> 
> 
> Or check if relay board is _active Low_ or _active High_

@imdhanush 
How can I get this whether its Active low or Active High?
Theoretically I knew this but practically I don't.
I have observed that all outputs are high (default) after Code uploading and as Default Power on.
[image]
I am going to construct a relay circuit for this project if there is not error is noticed all other aspects.
[image]

Kindly Suggest.
Thanks and regards

imdhanush, 2019-08-22:
Okay I checked the description for that relay channel board. It's a active Low board.

When you give input as LOW relay will be ON . When you give input as HiGH relay will be OFF.

amitabheer, 2019-08-22:
@imdhanush 

Thank you dear Dhanush Ji
It means problem is resolved. Kindly do me a favour that you check whether my relay circuit (photo shared i previous message) would be fine or some amendment is required in the same? and do share the description of the channel board so that I can get more information on the device.

Thanks & regards

imdhanush, 2019-08-22:
@amitabheer The relay board you are using now it's good. It has a optocoupler which offers maximum impedance from relay to NodeMCU. 

According to your relay schematic a transistor is used switch relay state. Incase if flyback diode fails or a high voltage is passed through transistor it will breakdown and will damage nodeMCU. The present board is reliable more. :)

imdhanush, 2019-08-22:
In your code .
Just change all HIGH's to LOW
And all LOW's to HIGH

Relay board description is here
http://wiki.sunfounder.cc/index.php?title=8_Channel_5V_Relay_Module

amitabheer, 2019-08-22:
> @amitabheer The relay board you are using now it's good. It has a optocoupler which offers maximum impedance from relay to NodeMCU.
>
> According to your relay schematic a transistor is used switch [...]

@iamdhanush
Thank you very much for your help. 
Further I need suggestion for making 12 V relay system with optocouplers. Actually I have 12V power supply there in every switchboard at my home and that is why I shall prefer to get power from there (sufficient current to drive the realy rather than converting into +5V for relays.) 
I shall obtain +5V using LM7805 and relay ratings will be kept at 12V.
Kindly give your frank opinion.
Best regards
-Dr Amit

amitabheer, 2019-08-22:
> 
> 
> In your code .
> Just change all HIGH's to LOW
> And all LOW's to HIGH
> 
> Relay board description is here
> http://wiki.sunfounder.cc/index.php?title=8_Channel_5V_Relay_Module

@imdhanush 
I have made the changes and uploaded the program to the Node MCU and the problem has been resolved but still by default all Relays are on (when the circuit is powered on).
Is this can be rectified via making changes in the program?
Kindly revert when you have time.
I appreciate your help.
Thanks & regards

imdhanush, 2019-08-22:
In void setup ()
After pinMode ()

Set all pins to high

digitalWrite(device_1,HIGH)
So on..

amitabheer, 2019-08-22:
> 
> 
> In void setup ()
> After pinMode ()
> 
> Set all pins to high
> 
> digitalWrite(device_1,HIGH)
> So on..

@imdhanush 

I have made the changes but didn't work at all. It seems still something left to act upon.
Further I would like to add one important feature to this prototype as under:
1.Blinking LED indicator for searching for wifi and constant indicator on wifi access and like.
2.I shall add more devices with other set of MCU circuits to control almost all switches of my house.

In addition please let me know how many pins except (5,4,0,2,14,12, 13 and 15) can be used for obtaining output.

I appreciate your time.
Thank you

imdhanush, 2019-08-22:
Thanks, 
Can I see what changes you made. Mail me the code dhanushdazz@gmail.com

emnik, 2019-08-22:
First of all regarding the schematic which is drawn above I think you have
to use a NPN transistor and the relay will be connected to it's collector.
Second you should have an optocoupler in the circuit. I'll have to find a
schematic to send you if you are interested.
Last regarding the initial state of your devices, the relays
1) as commented above you can set the relay inputs to low or high at setup
() or
2) the relays have a normal open and a normal close connection. So you can
decide where you want to connect each device. Either to the com (common)
and the relay port that has power initially or to the com and the other
port that doesn't have power initially.

amitabheer, 2019-08-22:
> First of all regarding the schematic which is drawn above I think you have to use a NPN transistor and the relay will be connected to it's collector. Second you should have an optocoupler in the [...]

@emnik 
Thank you for your reply
As per your response I request you to suggest me the schematic diagram to satisfy the need. You are also requested to make an optional and additional schematic that I can use for 12 V relays.
I appreciate your time and efforts. Thank you for helping me learning.
Best regards

estebanhirzfeld, 2019-09-15:
> Dear you,
> Greetings and thanks to your contribution here.
> Kindly acknowledge that I too have facing this kind of problem with NODE MCU Lolin V3.
> I have made a project on 8_Ch relay control via Alexa [...]

hi! i have the same board (LoLin v3 MCU) and a 8 Channel relay but i cant control with alexa, the relays are always on, i can detect the 8 devices but i cant control then, i use the switch_example from sinric master

amitabheer, 2019-09-16:
Dear You,

Can you get your MCU online??
You can check this at www.sinric.com. This can be seen at the bottom left
corner of the webpage under your devices listing there.
Kindly share your code to me at amit.mbacm@gmail.com if you want me to
check the same. (don't share your API Key)
Regards

estebanhirzfeld, 2019-09-17:
sent

estebanhirzfeld, 2019-09-22:
> Dear You, Can you get your MCU online?? You can check this at [www.sinric.com](http://www.sinric.com). This can be seen at the bottom left corner of the webpage under your devices listing there. [...]

hello?

jflapao, 2020-06-11:
Hi everyone, I have the same problem that @amitabheer , @estebanhirzfeld 
Can upload the code the nodeMCU is online, can see the action on the serial of the arduino IDE but the state of the relay remains the same.
I have try to declare the nodeMCU pins as 0, 2, etc, etc, or D0, D2, etc, etc and nothing seams that the action reach the nodeMCU, but don't pass to the relay board.
My board is the same that @amitabheer and I can´t see the problem, if somebody can help I with loved