kakopappa/sinric #329: ESP 8266 V3  <FS.h> WiFiManager.h

arduino-nour, 2019-08-24:
[image]
[image]
[image]
////////////////////////////////////////////////////////////////////
////////////////////////////////////////////////////////////////////
arduinou.nour@gmai.com
//////////////////////////////////////////////////////////////////
//////////////////////////////////////////////////////////////////

[0.0.txt](https://github.com/kakopappa/sinric/files/3537202/0.0.txt)
/////////////////////////////////////////////////////////////////

#include <FS.h>                   //this needs to be first, or it all crashes and burns...
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WebSocketsClient.h> //  get it from https://github.com/Links2004/arduinoWebSockets/releases 
#include <ArduinoJson.h> // get it from https://arduinojson.org/ or install via Arduino library manager
#include "WiFiManager.h"          //https://github.com/tzapu/WiFiManager

// Issue #18 https://github.com/kakopappa/sinric/issues/18
// If you receive an error like multi_nav_wifi:51: error: 'class WiFiManager' has no member named 'addAP' use
// https://github.com/the-real-orca/WiFiManager

WebSocketsClient webSocket;
WiFiClient client;

#define MyApiKey "xxxxxxxxxxxxxxxxxxxxxxx" // TODO: Change to your sinric API Key. Your API Key is displayed on sinric.com dashboard

#define HEARTBEAT_INTERVAL 300000 // 5 Minutes 

#define lamp1 "5d344e1b56393964a3c9ba0b"

uint64_t heartbeatTimestamp = 0;
bool isConnected = false;

void load(WiFiManager &wifiManager) {
  //clean FS, for testing
  //  SPIFFS.format();

  //read configuration from FS json
  Serial.println("mounting FS...");

  if (SPIFFS.begin()) {
    if (SPIFFS.exists("/config.json")) {
      //file exists, reading and loading
      Serial.println("reading config file");
      File configFile = SPIFFS.open("/config.json", "r");

      if (configFile) {
        Serial.println("opened config file...");
        size_t size = configFile.size();
        // Allocate a buffer to store contents of the file.
        std::unique_ptr<char[]> buf(new char[size]);

        configFile.readBytes(buf.get(), size);
        DynamicJsonBuffer jsonBuffer;
        JsonObject& json = jsonBuffer.parseObject(buf.get());
        json.printTo(Serial); // debug output to console
        
        if (json.success()) {
          Serial.println("\nparsed json");
          if ( json["Networks"].is<JsonArray>() ) {
            for (int i=0; i < json["Networks"].size(); i++) {
              auto obj = json["Networks"][i];

              // add existing networks and credentials
              wifiManager.addAP(obj["SSID"], obj["Password"]);
            }
[... 186 lines cut ...]
  webSocket.setReconnectInterval(5000);   // If you see 'class WebSocketsClient' has no member named 'setReconnectInterval' error update arduinoWebSockets
}

void loop() {
  webSocket.loop();
  
  if(isConnected) {
      uint64_t now = millis();
      
      // Send heartbeat in order to avoid disconnections during ISP resetting IPs over night. Thanks @MacSass
      if((now - heartbeatTimestamp) > HEARTBEAT_INTERVAL) {
          heartbeatTimestamp = now;
          webSocket.sendTXT("H");          
      }
  }   
}

//////////////////////////////////////////////////////////

Lod does not work
Even the ESP 8266 V3  does not light up when you press the key What is the problem?
Is it in the code?
The connection is correct
Thank you for your attention
/////////////////////////////////////////

arduino-nour, 2019-08-26:
????

kakopappa, 2019-08-26:
in the serial terminal pic, I can see the on/off commands coming in, looks like your Sinric integration si fine. 

If the LED is not working, it's likely to do with the PIN mapping is wrong. 

try the on.off on a new sketch with the same pins and check whether it's working

arduino-nour, 2019-08-26:
Thank you for your comment
All connections are correct
You changed the  esp 8266 v3
It's working on ESP8266WebServer library
Also on the WiFiManager.h library
But does not work with Alexa
Everything works successfully even if you log on Serial Monitor
Gives the code On and OFF and it sends the status of keys on
On the mobile, it also gives On and Off from within the Alexa program

Is the code correct?

arduino-nour, 2019-08-26:
Thank you for your comment
All connections are correct
You changed the  esp 8266 v3
It's working on ESP8266WebServer library
Also on the WiFiManager.h library
But does not work with Alexa
Everything works successfully even if you log on Serial Monitor
Gives the code On and OFF and it sends the status of keys on
On the mobile, it also gives On and Off from within the Alexa program

Is the code correct?

‫في الاثنين، 26 أغسطس 2019 في 2:28 م تمت كتابة ما يلي بواسطة ‪Aruna
Tennakoon‬‏ <‪notifications@github.com‬‏>:‬

arduino-nour, 2019-08-26:
It worked like this

#define lamp1 "5xxxxxxxxxxxxxxxxxxxxxxxx"
const int ligh1= 16;

void turnOn(String deviceId) {
  if (deviceId == "lamp1") // Device ID of first device
  {  
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    digitalWrite (ligh1,HIGH);
  }
void setup() {
  Serial.begin(115200);
  pinMode (ligh1,OUTPUT);
////////////////////////////////////////////////

Do you mean work?

oid turnOn(String deviceId) {
  if (deviceId == "5xxxxxxxxxxxxxxxx") // Device ID of first device
  {  
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    digitalWrite (ligh1,HIGH);
  }

////////////////////////////////////////////////
All thanks to you my friend

kakopappa, 2019-08-26:
if (deviceId == "lamp1") <-- wrong

this should be device id. device id is like 5xxxxxxxxxxxxxx

arduino-nour, 2019-08-27:
kakopappa
thank you very much
I will work it out and give you the result

kakopappa, 2019-08-27:
device id is not device name. I hope you got it to work now

arduino-nour, 2019-08-27:
@kakopappa 
thank you my friend
The problem has been resolved
Now I can call Alexa

oid turnOn(String deviceId) {
if (deviceId == "5xxxxxxxxxxxxxxxx") // Device ID of first device
{
Serial.print("Turn on device id: ");
Serial.println(deviceId);
digitalWrite (16,HIGH);
}

closed 2019-08-28 by kakopappa