kakopappa/sinric #330: Open SmartTV apps?

duartesaraiiva, 2019-08-24:
I'm trying to make my smart tv open apps with sinric and alexa (ex: netflix), but after digging into sinric code I can't find any trigger for that. Could you implement one trigger such as "startApp" or "closeApp" ?

kakopappa, 2019-08-25:
How do you make apps for smart tv? Which framework or tools do you use?

duartesaraiiva, 2019-08-25:
I'm using this library to establish a connection between my pc and my LG TV: https://github.com/klattimer/LGWebOSRemote/
There is a startApp and a closeApp that requires a appid. You could return the App name or the appId and everyone could implement according to their library.
The invocation could be like: “Alexa, open/start Netflix”
My whole project: https://github.com/FeLiNa22/amazon-alexa-lg-tv