kakopappa/sinric #331: Alexa discovered and added new devices. Also Sinric android app is showing the devices but on turning ON/OFF the device from alexa app no change in ESP8266 relays. But relay state is changing via Sinric app. Thus alexa although connected is not sending commands to Sinric. Please Help!!!

Hoshang1121, 2019-09-01:


kakopappa, 2019-09-01:
post the code here without api keys

Hoshang1121, 2019-09-01:
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <WebSocketsClient.h> //  https://github.com/kakopappa/sinric/wiki/How-to-add-dependency-libraries
#include <ArduinoJson.h> // https://github.com/kakopappa/sinric/wiki/How-to-add-dependency-libraries (use the correct version)
#include <StreamString.h>

ESP8266WiFiMulti WiFiMulti;
WebSocketsClient webSocket;
WiFiClient client;

#define HEARTBEAT_INTERVAL 300000 // 5 Minutes 

uint64_t heartbeatTimestamp = 0;
bool isConnected = false;
 
#define MyApiKey "" // TODO: Change to your sinric API Key. Your API Key is displayed on sinric.com dashboard
#define MySSID "" // TODO: Change to your Wifi network SSID
#define MyWifiPassword "" // TODO: Change to your Wifi network password

#define DEVICE1 ""   //TODO: Device ID of first device
#define DEVICE2 ""  //TODO: Device ID of second device
 
const int relayPin1 = 16; // TODO: Change according to your board
const int relayPin2 = 2; // TODO: Change according to your board

// deviceId is the ID assgined to your smart-home-device in sinric.com dashboard. Copy it from dashboard and paste it here

void turnOn(String deviceId) {
  if (deviceId == DEVICE1)
  {  
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    
     digitalWrite(relayPin1, HIGH); // turn on relay with voltage HIGH
  }
  else  if (deviceId == DEVICE2)
  {  
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    
     digitalWrite(relayPin2, HIGH); // turn on relay with voltage HIGH
  }  
  else {
    Serial.print("Turn on for unknown device id: ");
    Serial.println(deviceId);
  }     
}

void turnOff(String deviceId) {
   if (deviceId == DEVICE1)
   {  
     Serial.print("Turn off Device ID: ");
     Serial.println(deviceId);
     
     digitalWrite(relayPin1, LOW);  // turn off relay with voltage LOW
   }
   else if (deviceId == DEVICE2)
   {  
     Serial.print("Turn off Device ID: ");
     Serial.println(relayPin2);
     
     digitalWrite(relayPin2, LOW);  // turn off relay with voltage LOW
   }
   else {
     Serial.print("Turn off for unknown device id: ");
     Serial.println(deviceId);    
  }
}

void webSocketEvent(WStype_t type, uint8_t * payload, size_t length) {
  switch(type) {
    case WStype_DISCONNECTED:
      isConnected = false;    
      Serial.printf("[WSc] Webservice disconnected from sinric.com!\n");
      break;
    case WStype_CONNECTED: {
      isConnected = true;
      Serial.printf("[WSc] Service connected to sinric.com at url: %s\n", payload);
      Serial.printf("Waiting for commands from sinric.com ...\n");        
      }
      break;
    case WStype_TEXT: {
        Serial.printf("[WSc] get text: %s\n", payload);
        // Example payloads

        // For Switch or Light device types
        // {"deviceId": xxxx, "action": "setPowerState", value: "ON"} // https://developer.amazon.com/docs/device-apis/alexa-powercontroller.html

        // For Light device type
        // Look at the light example in github
          
        DynamicJsonBuffer jsonBuffer; 
        JsonObject& json = jsonBuffer.parseObject((char*)payload); 
        String deviceId = json ["deviceId"];     
        String action = json ["action"];
        
        if(action == "setPowerState") { // Switch or Light
            String value = json ["value"];
            if(value == "ON") {
                turnOn(deviceId);
            } else {
                turnOff(deviceId);
            }
        }
        else if (action == "test") {
            Serial.println("[WSc] received test command from sinric.com");
        }
      }
      break;
    case WStype_BIN:
      Serial.printf("[WSc] get binary length: %u\n", length);
      break;
  }
}

void setup() {
  Serial.begin(74880);
    
  pinMode(relayPin1, OUTPUT);
  pinMode(relayPin2, OUTPUT);
  
  WiFiMulti.addAP(MySSID, MyWifiPassword);
  Serial.println();
  Serial.print("Connecting to Wifi: ");
  Serial.println(MySSID);  

  // Waiting for Wifi connect
  while(WiFiMulti.run() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  if(WiFiMulti.run() == WL_CONNECTED) {
    Serial.println("");
    Serial.print("WiFi connected. ");
    Serial.print("IP address: ");
    Serial.println(WiFi.localIP());
  }

  // server address, port and URL
  webSocket.begin("iot.sinric.com", 80, "/");

  // event handler
  webSocket.onEvent(webSocketEvent);
  webSocket.setAuthorization("apikey", MyApiKey);
  
  // try again every 5000ms if connection has failed
  webSocket.setReconnectInterval(5000);   // If you see 'class WebSocketsClient' has no member named 'setReconnectInterval' error update arduinoWebSockets
}

void loop() {
  webSocket.loop();
  
  if(isConnected) {
      uint64_t now = millis();
      
      // Send heartbeat in order to avoid disconnections during ISP resetting IPs over night. Thanks @MacSass
      if((now - heartbeatTimestamp) > HEARTBEAT_INTERVAL) {
          heartbeatTimestamp = now;
          webSocket.sendTXT("H");          
      }
  }   
}

kakopappa, 2019-09-01:
Code looks correct. 

1. FYI The device id is not the same device name. Sometimes people mistake id for the name. Check this part
Device ID is like 5aae40f2103b5554780exxxxx. Make sure your DEVICE1 and DEVICE2 is correct

2. Do you see commands in the Serial Monitor when you ask Alexa to turn on/off the device?

Hoshang1121, 2019-09-01:
Thanks for the reply sir. My device id is 5d695b611bddb66e363da747 and serial monitor is showing output on turning on device via Sinric app but no output is there when ON/OFF is through Alexa app. Although alexa is connected to Sinric and showing the device 1 (i.e. Light 1) ON/OFF state change on its UI.

kakopappa, 2019-09-01:
If you are not see anything in the serial monitor when you ask to turn on/off via Alexa then something else is wrong,

1. delete all device from sinric. 
2. got to alexa app and remove all device
3. create a new device in sinric. (dont forget to put the device description)
4. use the empty sinric connection template to connect to sinric server. (dont add relay code)
5. run the discovery in Alexa. 
6. try on-off command

Hoshang1121, 2019-09-01:
use the empty sinric connection template to connect to sinric server. (dont add relay code)?

You mean turnON() and turnOFF() functions? Without these what commands will be sent to esp8266?

kakopappa, 2019-09-01:
Use this
https://github.com/kakopappa/sinric/blob/master/arduino_examples/switch_example.ino

change the API KEY, WIFI Details only. Should should be able to see the commands coming from Alexa.

Hoshang1121, 2019-09-01:
Tried it still same issue. Very amazed to see why nothing is happening? 

Also on adding device in alexa it is automatically naming it Light 1 and Light 2 although in sinric names are Room and Kitchen. The device description is also showing not set in Alexa. Though description is set to Bedroom light in Sinric. Is it okay?

kakopappa, 2019-09-02:
This is strage. Not sure what's going on..

Is your API Key correct ? Do you have multiple accounts? try creating a new account

Steve-Broderick, 2019-09-02:
The JSON code needs to be upgraded to version 6. JSON.BUFFER is no longer supported.

Hoshang1121, 2019-09-02:
But the Sinric app is sending data to Esp8266 which means Json encode/decode parsing action is working fine. The issue seems to be at a bridge between Alexa and Sinric. I dont think the JSON has any role at the bridge between Alexa and Sinric.

Steve-Broderick, 2019-09-03:
I don't want to get off topic, but, you are correct, JSON is only the parser of the information sent back from Sinric. If I use your example it will not compile with JSON 6.x

JSON was recently updated to version 6 to make it 'easier' to parse. Version 5 code with the new library's does not work anymore.

They made some nice improvements in version 6 but the code is different. Here is a link to the migration path.  https://arduinojson.org/v6/doc/upgrade/

kakopappa, 2019-09-04:
Seems like when alexa sends a request, the server cannot find your account to deliver the message. Do you have more than 1 sinric account ?

> Also on adding device in alexa it is automatically naming it Light 1 and Light 2 althought in sinric names are Room and Kitchen. The device description is also showing not set in Alexa. Though description is set to Bedroom light in Sinric.

Looks like your account and Alexa skill is not in sync. Best solution would be to create a new account. It doesn't have to be a real email address, you can try with a dummy