kakopappa/sinric #335: ESP Disconnects over Time

SiddhantShah2001, 2019-09-10:
**Hey there!** 
Thank You for creating such a great Library/Platform. I really Appreciate your Efforts.

So here the Problem is that my ESP8266-module/**NodeMCU disconnects from the Server randomly**. To Revive the Connection i have to manually press the Reset Button on the Board or Turn of the Power Supply. Other way to Reconnect is to Restart the Router.

There are 2 Ways to Disconnect: i.e WiFi itself(Router) or the Server(Modem).
01. WiFi: If the WiFi signal is lost or basically the Router is turned Off, the Esp Module Displays 
      Error Messages through the Serial Monitor (as it should).
      Disconnecting and Reconnecting the WiFi makes the Esp connect back to the Server.
 but,
02. Server: If the Server is Disconnected, no Error Messages are Shown by the Esp Module. 
                  Disconnecting and reconnecting the Modem to the Router doesn't help in this case.

I have also used other Libraries like FauxmoESP and EspAlexa. They work Really Great and Handle the Disconnections very well (maybe because they are local servers).

But i find Sinric to be more Vast/Advanced and Easy to use. I am planning to do Full Home Automation but my Work's at a halt as:
01. Other Libraries have no Options to change the Device Type from "Light", as i have Ceiling 
      Fans included in the Automation and i need them to be "Switch Devices" and not "Light 
      devices" because for example if i say "Turn Off All Lights" it will turn off the Fans as well.
02. These other Libraries have Discovery issues which are really really Annoying. Considering to 
      Automate more than 30 or 40 Appliances across the House it's going to be Super 
      Cumbersome setting them up all. (and if by mistake any device is deleted, Re-discovering 
      it will be a pain in the ass)

Thank you for your valuable time! 
Hoping for a Positive Reply.

kakopappa, 2019-09-10:
Hi there. Thanks 

disconnection detection is hard. It takes time for TCP/IP ACK to complete and close the connection properly. Sometime it's the router, sometimes it's the ISP, sometimes it's the sinric server resetting the connection. Best option is to detect disconnections in the client side and reconnect. Have you tried enabling websocket heartbeat ?
https://github.com/Links2004/arduinoWebSockets/blob/master/src/WebSocketsClient.h#L91

This tend to improve the disconnect /  reconnect detection. Also make sure setReconnectInterval is set

Other libraries like FauxmoESP and EspAlexa they emulate a belkin device or a hue device, I do not think this is the right thing to do. That's why i created Sinric.

SiddhantShah2001, 2019-09-15:
Yeah i think the Heartbeats are enabled, not sure!
I used the Example Sketch and just modified the Id and Stuff. Not sure if there's anything else to be done to Establish Heartbeat.

Oh and by the way i heard(read) about Sinic 2.0 in some other thread. Could you share some more details about it if it's not a problem ?!
Like, 
What would be the new features in it ? 
When is it likely to be released ?
and stuff like that in detail.

Sorry for such a late reply.

ajarsco, 2020-12-01:
Any updates! I am facing the same issue
The board connects for a long time... but after a while the board losses connection and fails to reconnect..
But in normal working times, the board reconnects to the wifi automatically. After long connections only the board fails to reconnect.

Kindly reply anyone faster