kakopappa/sinric #336: DynamicJsonBuffer error

Likhith152, 2019-09-11:
I'm getting this DynamiJsonBuffer error. Tried to upgrade from v5 to v6, but no use. Will you please help me out in this topic,. thanks
[image]

kakopappa, 2019-09-11:
you have two options

1. downgrade to 5
2 change the code to support 6
https://arduinojson.org/v6/doc/upgrade/

Likhith152, 2019-09-12:
Sorry,
Will you please change the code to support 6 for me!
Thanks in advANCE

Likhith152, 2019-09-12:
And now Im getting this error
'ArduinoJson6120_000001::DynamicJsonDocument' has no member named 'parseObject'

[image]

imdhanush, 2019-09-12:
Current websocket library works fine with 5th verison of JSON library. So it's better to downgrade to 5 .

Likhith152, 2019-09-14:
> you have two options
> 
> 1. downgrade to 5
>    2 change the code to support 6
>    https://arduinojson.org/v6/doc/upgrade/

Worked both. Thamks a lot for the support

Likhith152, 2019-09-14:
Thanks guys.
Tried  both ways. Downgrading the ArduinoJson library & Upgrading the code to v6 by manual. Both worked fine.

closed 2019-09-14 by Likhith152

MacSass, 2019-09-14:
@Likhith152 
I´d like to use upgraded code too - would you mind sharing your upgraded sample code, as I´m not too good with coding ...

Thanks - MacSass

PS: Would be nice if @kakopappa could upgrade the examples to v6 as well, so people don´t fall this trap when they try sinric ...

Likhith152, 2019-09-14:
//

//[sinric_v6.zip](https://github.com/kakopappa/sinric/files/3612689/sinric_v6.zip)

Give a try here
/*
 Version 0.4 - April 26 2019
*/ 

#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <WebSocketsClient.h> //  https://github.com/kakopappa/sinric/wiki/How-to-add-dependency-libraries
#include <ArduinoJson.h> // https://github.com/kakopappa/sinric/wiki/How-to-add-dependency-libraries (use the correct version)
#include <StreamString.h>

ESP8266WiFiMulti WiFiMulti;
WebSocketsClient webSocket;
WiFiClient client;

#define MyApiKey "" // TODO: Change to your sinric API Key. Your API Key is displayed on sinric.com dashboard
#define MySSID "" // TODO: Change to your Wifi network SSID
#define MyWifiPassword "" // TODO: Change to your Wifi network password

#define HEARTBEAT_INTERVAL 300000 // 5 Minutes 

uint64_t heartbeatTimestamp = 0;
bool isConnected = false;

// deviceId is the ID assgined to your smart-home-device in sinric.com dashboard. Copy it from dashboard and paste it here

void turnOn(String deviceId) {
  if (deviceId == "5axxxxxxxxxxxxxxxxxxx") // Device ID of first device
  {  
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
  } 
  else if (deviceId == "5axxxxxxxxxxxxxxxxxxx") // Device ID of second device
  { 
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
  }
  else {
    Serial.print("Turn on for unknown device id: ");
    Serial.println(deviceId);    
  }     
}

void turnOff(String deviceId) {
   if (deviceId == "5axxxxxxxxxxxxxxxxxxx") // Device ID of first device
   {  
     Serial.print("Turn off Device ID: ");
     Serial.println(deviceId);
   }
   else if (deviceId == "5axxxxxxxxxxxxxxxxxxx") // Device ID of second device
   { 
     Serial.print("Turn off Device ID: ");
     Serial.println(deviceId);
  }
  else {
     Serial.print("Turn off for unknown device id: ");
     Serial.println(deviceId);    
  }
}

void webSocketEvent(WStype_t type, uint8_t * payload, size_t length) {
  switch(type) {
    case WStype_DISCONNECTED:
      isConnected = false;    
      Serial.printf("[WSc] Webservice disconnected from sinric.com!\n");
      break;
    case WStype_CONNECTED: {
      isConnected = true;
      Serial.printf("[WSc] Service connected to sinric.com at url: %s\n", payload);
      Serial.printf("Waiting for commands from sinric.com ...\n");        
      }
      break;
    case WStype_TEXT: {
        Serial.printf("[WSc] get text: %s\n", payload);
        // Example payloads

        // For Switch or Light device types
        // {"deviceId": xxxx, "action": "setPowerState", value: "ON"} // https://developer.amazon.com/docs/device-apis/alexa-powercontroller.html

        // For Light device type
        // Look at the light example in github
          
       DynamicJsonDocument doc(200);
        deserializeJson(doc, (char*)payload); 
        String deviceId = doc ["deviceId"];     
        String action = doc ["action"];
        
        if(action == "setPowerState") { // Switch or Light
            String value = doc ["value"];
            if(value == "ON") {
                turnOn(deviceId);
            } else {
                turnOff(deviceId);
            }
        }
        else if (action == "SetTargetTemperature") {
            String deviceId = doc ["deviceId"];     
            String action = doc ["action"];
            String value = doc ["value"];
        }
        else if (action == "test") {
            Serial.println("[WSc] received test command from sinric.com");
        }
      }
      break;
    case WStype_BIN:
      Serial.printf("[WSc] get binary length: %u\n", length);
      break;
  }
}

void setup() {
  Serial.begin(115200);
  
  WiFiMulti.addAP(MySSID, MyWifiPassword);
  Serial.println();
  Serial.print("Connecting to Wifi: ");
  Serial.println(MySSID);  

  // Waiting for Wifi connect
  while(WiFiMulti.run() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  if(WiFiMulti.run() == WL_CONNECTED) {
    Serial.println("");
    Serial.print("WiFi connected. ");
    Serial.print("IP address: ");
    Serial.println(WiFi.localIP());
  }

  // server address, port and URL
  webSocket.begin("iot.sinric.com", 80, "/");

  // event handler
  webSocket.onEvent(webSocketEvent);
  webSocket.setAuthorization("apikey", MyApiKey);
  
  // try again every 5000ms if connection has failed
  webSocket.setReconnectInterval(5000);   // If you see 'class WebSocketsClient' has no member named 'setReconnectInterval' error update arduinoWebSockets
}

void loop() {
  webSocket.loop();
  
  if(isConnected) {
      uint64_t now = millis();
      
      // Send heartbeat in order to avoid disconnections during ISP resetting IPs over night. Thanks @MacSass
      if((now - heartbeatTimestamp) > HEARTBEAT_INTERVAL) {
          heartbeatTimestamp = now;
          webSocket.sendTXT("H");          
      }
  }   
}

// If you want a push button: https://github.com/kakopappa/sinric/blob/master/arduino_examples/switch_with_push_button.ino

MacSass, 2019-09-15:
Hey, thanks - I´ll take a look ...

MacSass, 2019-12-23:
Hello,
I finally found the time to update my code to v6 using the code samples from @Likhith152  - works great and easy enough when you know what you need to change.

Thanks a lot for the sample code - really appreciated.

I would recommend to change all code samples to v6, so new users have it more easy to get them running.

Regards - MacSass

sivar2311, 2019-12-23:
All Arduino examples (except multi_wifi) are compatible to ArduinoJson 5 and 6 since November 28.
Have a look at the new example code.

MacSass, 2019-12-23:
Wow - yes, just saw this now ...
Sorry I wasn´t following closely lately due to work issues - great job - nice.

Thanks MacSass