kakopappa/sinric #340: integrate a piece of code for a stepper motor

diepie, 2019-09-16:
In the Sinric Switch sample code ( switch_example.ino ->  https://github.com/kakopappa/sinric/blob/master/arduino_examples/switch_example.ino )
I have to integrate a piece of code for a stepper motor  that needs to loop,  after device ID 1 is true, so after this line:

void turnOn(String deviceId) {
  if (deviceId == "5cf8de1d0837cc3a87c8faf5") // Device ID of first device

This Is The piece of code that has to loop: 

void loop() {

  //Change direction when the stepper reaches the target position
  if (stepper1.distanceToGo() == 0) {
    stepper1.moveTo(-stepper1.currentPosition());
  }
  stepper1.run();
}

Question Is Where do I have to put this  peace of code so that it loops, i cannot have to instances of void loop()

Regards Peter

kakopappa, 2019-09-16:
what are trying to do ? turn on the stepper motor? You can't have a loop inside void turnOn(String deviceId) ...

I dont understand what are you tyring to do here. So i am going to guess based on a command you receive, you are trying to move the stepper

bool moveStepper = false;

void turnOn(String deviceId) {
   if (deviceId == "5cf8de1d0837cc3a87c8faf5") // Device ID of first device
   {
       moveStepper  = true;
   }
}

void loop() {
    ....
   if(moveStepper  == true) {
     moveStepper  = false;
 
if (stepper1.distanceToGo() == 0) {
stepper1.moveTo(-stepper1.currentPosition());
}
stepper1.run();
}
  
   }
}

diepie, 2019-09-16:
thanks, i'm going to try and come back to it soon

diepie, 2019-09-16:
The complete code with your proposal is pasted below.
The problem is still that the piece of code that the stepper motor has to run is only executed once when it is switched on. This code must loop until the current Position is reached

#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <WebSocketsClient.h> //  https://github.com/kakopappa/sinric/wiki/How-to-add-dependency-libraries
#include <ArduinoJson.h> // https://github.com/kakopappa/sinric/wiki/How-to-add-dependency-libraries
#include <StreamString.h>

#include <AccelStepper.h>
#define HALFSTEP 8

ESP8266WiFiMulti WiFiMulti;
WebSocketsClient webSocket;
WiFiClient client;

#define MyApiKey "4983cfee-24ef-4b82-b0b1-9dc43ac9e057" // TODO: Change to your sinric API Key. Your API Key is displayed on sinric.com dashboard
#define MySSID "diepie_2.4" // TODO: Change to your Wifi network SSID
#define MyWifiPassword "hcbq97ah" // TODO: Change to your Wifi network password

// Motor pin definitions
#define motorPin1  5    // IN1 on the ULN2003 driver 1
#define motorPin2  4     // IN2 on the ULN2003 driver 1
#define motorPin3  0     // IN3 on the ULN2003 driver 1
#define motorPin4  2    // IN4 on the ULN2003 driver 1

// Initialize with pin sequence IN1-IN3-IN2-IN4 for using the AccelStepper with 28BYJ-48
AccelStepper stepper1(HALFSTEP, motorPin1, motorPin3, motorPin2, motorPin4);

#define HEARTBEAT_INTERVAL 300000 // 5 Minutes 

const int relayPin = D1;

uint64_t heartbeatTimestamp = 0;
bool isConnected = false;

void setPowerStateOnServer(String deviceId, String value);
void setTargetTemperatureOnServer(String deviceId, String value, String scale);

// deviceId is the ID assgined to your smart-home-device in sinric.com dashboard. Copy it from dashboard and paste it here

bool moveStepper = false;

void  turnOn(String deviceId) {
  if (deviceId == "5cf8de1d0837cc3a87c8faf5") // Device ID of first device
  {
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);

    digitalWrite(relayPin, HIGH); // turn on relay with voltage HIGH

    moveStepper = true;

  }
  else {
    Serial.print("Turn on for unknown device id: ");
    Serial.println(deviceId);
  }
}

void turnOff(String deviceId) {
  if (deviceId == "5cf8de4d0837cc3a87c8faf7") // Device ID of first device
  {
    Serial.print("Turn off Device ID: ");
    Serial.println(deviceId);

    digitalWrite(relayPin, LOW);  // turn off relay with voltage LOW
  }
  else {
[... 129 lines cut ...]
  StreamString databuf;
  root.printTo(databuf);

  webSocket.sendTXT(databuf);
}

//eg: setPowerStateOnServer("deviceid", "CELSIUS", "25.0")

// Call ONLY If status changed. DO NOT CALL THIS IN loop() and overload the server.
void setTargetTemperatureOnServer(String deviceId, String value, String scale) {
  DynamicJsonBuffer jsonBuffer;
  JsonObject& root = jsonBuffer.createObject();
  root["action"] = "SetTargetTemperature";
  root["deviceId"] = deviceId;

  JsonObject& valueObj = root.createNestedObject("value");
  JsonObject& targetSetpoint = valueObj.createNestedObject("targetSetpoint");
  targetSetpoint["value"] = value;
  targetSetpoint["scale"] = scale;

  StreamString databuf;
  root.printTo(databuf);

  webSocket.sendTXT(databuf);
}

JanB97, 2019-10-10:
Hi,
without reading your whole code...
Remove "moveStepper = false;"  after "if(moveStepper == true) {" and put it in an if statement that only gets executed after the posisition is reached.
The loop will run until "moveStepper" is false, so don't set it to false if you want to move the stepper further.

This really has nothing to do with sinric though, it's simple arduino programming, you would be better of posting this in an arduino forum.