kakopappa/sinric #343: Upgrade existing Arduino examples to ArduinoJson 6

kakopappa, 2019-09-23:
Upgrade existing Arduino examples to ArduinoJson 6

Steve-Broderick, 2019-09-24:
Hi,

I have been using JSON 6 with Sinric for a while; how can I help?

kakopappa, 2019-09-24:
can you fork the repo, upgrade the arduino examples to  ArduinoJson 6. commit and send a pull request ?

Steve-Broderick, 2019-09-24:
I have no clue what you said. I sent you a message on facebook with my e mail address.

Steve-Broderick, 2019-09-24:
what does 'fork the repo' mean?

kakopappa, 2019-09-24:
it's okey. if you are new to git. 

just change the code to support ArduinoJson 6 and upload it somewhere and send me the link.

Thanks

Steve-Broderick, 2019-09-25:
OK,  I will make an example of a simple switch that has version 5 & 6 code in it. Uncomment the one you want to use.

Do you have a 'test' account that I can use to test these with or do I have to use my own account and change the AP, device ID's  etc keys before uploading.

Steve-Broderick, 2019-09-25:
Hi,  New to github, I made a simple example below that works with both JSON V5 and V6 depending on commented section.

It is quite verbose; I could have made it more concise but left some debug code in as examples.

Feel free to change this anyway you want and let me know if this is what you were looking for.

/* 
 * The below is a very simple example with some debug code for reference and examples
 * This sketch can work with JSON version 5 or JSON version 6 depending on what section you comment out 
 * in the "// JSON SECTION" just search for JSON SECTION and comment out the JSON version you don't want to use. 
 * Either, but not both, JSON sections must be uncommented.
 * 
 * This sketch was verified with JSON Version 5.13.5 and version 6.12.0
 * 
 * https://arduinojson.org/?utm_source=meta&utm_medium=library.properties
 * 
 * This sketch does not require a real device attached to it other than the 8266 of course, it was done with a WEMOS MINI D1 R1 
 * but should work with any ESP8266; in a real world application you would be controlling something like a swirch, light etc..
 * 
 * Agin, to try and make this simple I used the built in LED on the 8266 and you can turn that on using  an AMAZON ECHO device; 
 * I do not have a GOOGLE Home device.
 * 
 * The intent of this simple sketch is only to show JSON version 5 & 6 differences, nothin more.
 * 
 * look for the TODO: sections and change them accordingly; there are only a few things that need to be changed to work with 
 * your environment.
 * 
 */

   
#include <WebSockets.h>  // use version 2.1.0 later versions did not work for me
#include "WebSocketsClient.h" //  get it from https://github.com/Links2004/arduinoWebSockets/releases 
#include <WebSocketsServer.h>

#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <ArduinoJson.h> // get it from https://arduinojson.org/ or install via Arduino library manager
#include <StreamString.h>

ESP8266WiFiMulti WiFiMulti;
WebSocketsClient webSocket;
WiFiClient client;

// TODO: Change to your sinric API Key. Your API Key is displayed on sinric.com dashboard

#define MyApiKey "your API Key" 

// TODO: Change to your sinric device ID make it a switch type device!

#define TEST_DEVICE_DEV_ID "Your Device Key"  // from Sinric.com dashboard

// TODO: Change to the GPIO pin # used to control device, like a relay or something

#define TEST_DEVICE_GPIO_PIN  2  //GPIO Pin # (LED in this example)

// TODO: Change to your Wifi network SSID and password
#define MyWifiPassword "SSID PWD"
#define MySSID "YOUR SSID" 

#define HEARTBEAT_INTERVAL 300000 // 5 Minutes 

/* 

Don't forget to set PINMODE ! 

D1 Mini Map

Also if ian 8266 locks up try this:

[... 206 lines cut ...]
  // I chabged the above to see if it wouyld print the URL
  webSocket.begin("iot.sinric.com", 80, "/iot.sinric.com"); //"iot.sinric.com", 80

  // event handler
  webSocket.onEvent(webSocketEvent);
  webSocket.setAuthorization("apikey", MyApiKey);
  
  // try again every 5000ms if connection has failed
  webSocket.setReconnectInterval(5000);   // If you see 'class WebSocketsClient' has no member named 'setReconnectInterval' error update arduinoWebSockets
}

void loop() {
  
  webSocket.loop();

  if(isConnected) {
      uint64_t now = millis();
      
      // Send heartbeat in order to avoid disconnections during ISP resetting IPs over night. Thanks @MacSass
      if((now - heartbeatTimestamp) > HEARTBEAT_INTERVAL) {
          heartbeatTimestamp = now;
          webSocket.sendTXT("H");          
      }
  }   
}

kakopappa, 2019-09-25:
thanks. i will take a look

Steve-Broderick, 2019-09-26:
Hi,  I could not get your switch example to compile;  I do not see where 'action' was ever declared so I added these two line to your example; updated it to version 6 and it compiles now.

Look for this in your example and change it to the code I put below and it will compile  SPB are my initials.

        // Look at the light example in github

          /* Version 5
        DynamicJsonBuffer jsonBuffer;
        JsonObject& json = jsonBuffer.parseObject((char*)payload); 
        String deviceId = json ["deviceId"];     
        String action = json ["action"];
        */

        /* version 6 change */

DynamicJsonDocument json(1024);
      // get the root object is not necessary I left it here as an example
      JsonObject root = json.to<JsonObject>();       
      deserializeJson (json, (char*)payload);

      
        /* EO Version 6 change */

        /* added by SPB since 'action was never declared ! */
                
      String deviceId = json ["deviceId"];     
      String action = json ["action"];

/* EO my additions for action */
        if(action == "setPowerState") { // Switch or Light