kakopappa/sinric #344: libraries versions

emnik, 2019-09-27:
Hi! can please someone give me the working versions for the libraries used for
Websockets
Pubsubclient and
ArduinoOTA 
that use for
1. esp8266 and
2. esp32 ?
as the last days I can not connect with sinric.
I had no problems for quite a long time but trying to use esp32 I updates all the libraries and I cannot connect anymore.
OTA and MQTT works but the websocket to iot.sinric.com does not connect anymore neither with esp8266 nor with the esp32 :-(
I believe a library version is to blame as I have not change anything in my code but I cannot find which one...

kakopappa, 2019-09-27:
Websockets: https://github.com/Links2004/arduinoWebSockets/releases/tag/2.1.1 
JSON: https://github.com/bblanchon/ArduinoJson/releases/tag/v5.13.2 

not sure about others

emnik, 2019-09-27:
Can please someone test the basic code below? I use the 2.1.1 websockets library
and JSON v6.11.3 Everything used to work... but now I get [WSc] Webservice disconnected from sinric.com! every 5 seconds (the reconnect interval...)
`
#include <WebSocketsClient.h> //  get it from https://github.com/Links2004/arduinoWebSockets/releases 
#include <ArduinoJson.h> // get it from https://arduinojson.org/ or install via Arduino library manager SOS Version 6

WebSocketsClient webSocket;

const char* MyApiKey = "xxxxxxxxxxxxxxxxxxxxxxx"; // TODO: Change to your sinric API Key. Your API Key is displayed on sinric.com dashboard
const char* MySSID = "xxxxxx"; // TODO: Change to your Wifi network SSID
const char* MyWiFiPassword = "xxxxxxxxxxx"; // TODO: Change to your Wifi network password
const String MyDeviceID = "xxxxxxxxxxxxxxxxx"; // From sinric

// Static IP address configuration
IPAddress staticIP(192, 168, 1, 63); //ESP static ip
IPAddress gateway(192, 168, 1, 1);   //IP Address of your WiFi Router (Gateway)
IPAddress subnet(255, 255, 255, 0);  //Subnet mask

#define HEARTBEAT_INTERVAL 300000 // 5 Minutes 

const int RELAY = 22; // gpio 05 = pin D1 digital pin (without PWM) for relay control (also connected with the on NodeMCU.0.9 board blue led)

uint64_t heartbeatTimestamp = 0;
bool isConnected = false;

void setPowerStateOnServer(String deviceId, String value) {
  char databuf[128];
  StaticJsonDocument<128> root;
  root["deviceId"] = deviceId;
  root["action"] = "setPowerState";
  root["value"] = value;
  serializeJson(root, databuf);
  webSocket.sendTXT(databuf);
}

//Define the commands for device control
void turnOn(String deviceId) {
  if (deviceId == MyDeviceID) // Device ID of first device
  {
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    digitalWrite(RELAY, LOW); //turnOn
    setPowerStateOnServer(deviceId, "ON");
  }
}

void turnOff(String deviceId) {
  if (deviceId == MyDeviceID) // Device ID of first device
  {
    Serial.print("Turn off Device ID: ");
    Serial.println(deviceId);
    digitalWrite(RELAY, HIGH); //turnOff
    setPowerStateOnServer(deviceId, "OFF");
  }
}

void webSocketEvent(WStype_t type, uint8_t * payload, size_t length) {
  switch (type) {
    case WStype_DISCONNECTED:
      isConnected = false;
      Serial.printf("[WSc] Webservice disconnected from sinric.com!\n");
      break;
    case WStype_CONNECTED: {
        isConnected = true;
        Serial.printf("[WSc] Service connected to sinric.com at url: %s\n", payload);
        Serial.printf("Waiting for commands from sinric.com ...\n");
      }
      break;
    case WStype_TEXT: {
        Serial.printf("[WSc] get text: %s\n", payload);
        // Example payloads

        // For Switch or Light device types
        // {"deviceId": xxxx, "action": "setPowerState", value: "ON"} // https://developer.amazon.com/docs/device-apis/alexa-powercontroller.html

        // For Light device type
        // {"deviceId": xxxx, "action": "AdjustBrightness", value: 3} // https://developer.amazon.com/docs/device-apis/alexa-brightnesscontroller.html
        // {"deviceId": xxxx, "action": "setBrightness", value: 42} // https://developer.amazon.com/docs/device-apis/alexa-brightnesscontroller.html
        // {"deviceId": xxxx, "action": "SetColor", value: {"hue": 350.5,  "saturation": 0.7138, "brightness": 0.6501}} // https://developer.amazon.com/docs/device-apis/alexa-colorcontroller.html
        // {"deviceId": xxxx, "action": "DecreaseColorTemperature", value: 0} // https://developer.amazon.com/docs/device-apis/alexa-colortemperaturecontroller.html
        // {"deviceId": xxxx, "action": "IncreaseColorTemperature", value: 0} // https://developer.amazon.com/docs/device-apis/alexa-colortemperaturecontroller.html
        // {"deviceId": xxxx, "action": "SetColorTemperature", value: 2200} // https://developer.amazon.com/docs/device-apis/alexa-colortemperaturecontroller.html

        DynamicJsonDocument json(1024);
        deserializeJson(json, (char*)payload);
        String deviceId = json ["deviceId"];
        String action = json ["action"];

        if (action == "setPowerState") { // Switch or Light
          String value = json ["value"];
          if (value == "ON") {
            turnOn(deviceId);
          } else {
            turnOff(deviceId);
          }
        }
        else if (action == "test") {
          Serial.println("[WSc] received test command from sinric.com");
        }
      }
      break;
    case WStype_BIN:
      Serial.printf("[WSc] get binary length: %u\n", length);
      break;
  }
}

void setup() {
  Serial.begin(115200);
  pinMode(RELAY, OUTPUT);
  WiFi.mode(WIFI_STA);
  WiFi.config(staticIP, subnet, gateway);
  WiFi.begin(MySSID, MyWiFiPassword);
  uint64_t Uptime = millis();
  while (WiFi.waitForConnectResult() != WL_CONNECTED) {
    Serial.println("Connection Failed! Retrying in 1 sec...");
    delay(1000);
  }
  Serial.println("Ready");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  // server address, port and URL
  webSocket.begin("iot.sinric.com", 80, "/");

  // event handler
  webSocket.onEvent(webSocketEvent);
  webSocket.setAuthorization("apikey", MyApiKey);

  // try again every 5000ms if connection has failed
  webSocket.setReconnectInterval(5000);   // If you see 'class WebSocketsClient' has no member named 'setReconnectInterval' error update arduinoWebSockets
  //setPowerStateOnServer(MyDeviceID, "ON");
}

void loop() {
  //Handle websocket connections
  webSocket.loop();
  if (isConnected) {
    uint64_t now = millis();
    // Send heartbeat in order to avoid disconnections during ISP resetting IPs over night. Thanks @MacSass
    if ((now - heartbeatTimestamp) > HEARTBEAT_INTERVAL) {
      heartbeatTimestamp = now;
      webSocket.sendTXT("H");
    }
  }
}
`

emnik, 2019-09-27:
I found the issue. I had static IPs but I had not defined a DNS server! 
So while I had DHCP everything worked fine. After the static IPs new configuration, although the esps were connecting to the network they could not resolve the iot.sinric.com IP reporting a DNS lookup error -5 in the debug mode of websockets.
Setting a dns server for the static ip configuration solved the problem and everything is up again!

closed 2019-09-27 by emnik

kakopappa, 2019-09-28:
Thanks for the explanation. Good to know

why do you need to specify a DNS server? I don’t get it. Do you have some
kind of special network going on there?

emnik, 2019-09-28:
This is a good question. No I don't have any special network. Furthermore I works fine when on DHCP.
Now when I switched to Static IPs, with:
```
// Static IP address configuration
IPAddress staticIP(192, 168, 1, 63); //ESP static ip
IPAddress gateway(192, 168, 1, 1);   //IP Address of your WiFi Router (Gateway)
IPAddress subnet(255, 255, 255, 0);  //Subnet mask
```
and:
```
  WiFi.mode(WIFI_STA);
  WiFi.config(staticIP, gateway, subnet);
  WiFi.begin(MySSID, MyWiFiPassword);
```
I got (every 5 seconds:) 
`[WSc] Webservice disconnected from sinric.com!`
Then I enabled the debug option in WebSockets.h (#define DEBUG_ESP_PORT Serial) and in Arduino IDE (Tools>Debug Port > Serial & Tools> Debug Level > WiFi) and run the sketch again. This time I got:
```
[WS-Client] connect ws...
[hostByName] request IP for: iot.sinric.com
[hostByName] Host: iot.sinric.com lookup error: -6! // sometimes error: -5!
[WS-Client] connection to iot.sinric.com:80 Faild
[WS-Client] client disconnected.
[WSc] Webservice disconnected from sinric.com!
```
So I suspected a dns lookup error.
Adding the DNS option:
```
IPAddress staticIP(192, 168, 1, 63); //ESP static ip
IPAddress gateway(192, 168, 1, 1);   //IP Address of your WiFi Router (Gateway)
IPAddress subnet(255, 255, 255, 0);  //Subnet mask
IPAddress dns(1, 1, 1, 1);  //DNS This can also be IPAddress dns(192, 168, 1, 1)
```
and
```
  WiFi.mode(WIFI_STA);
  WiFi.config(staticIP, gateway, subnet, dns);
  WiFi.begin(MySSID, MyWiFiPassword);
```
Then everything worked and the debug messages were:
```
IP address: 192.168.1.63
[WS-Client] connect ws...
[hostByName] request IP for: iot.sinric.com
[hostByName] Host: iot.sinric.com IP: 198.23.139.123
[WS-Client] connected to iot.sinric.com:80.
[WS-Client][sendHeader] sending header...
[WS-Client][sendHeader] handshake GET / HTTP/1.1
Host: iot.sinric.com:80
Connection: Upgrade
Upgrade: websocket
Sec-WebSocket-Version: 13
Sec-WebSocket-Key: xxxxxxxxxxxxxxxxxxxxxx
Sec-WebSocket-Protocol: arduino
Origin: file://
User-Agent: arduino-WebSocket-Client
Authorization: Basic xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx

[write] n: zu t: 327
[WS-Client][sendHeader] sending header... Done (30902us).
[WS-Client][handleHeader] RX: HTTP/1.1 101 Switching Protocols
[WS-Client][handleHeader] RX: Server: nginx/1.10.3 (Ubuntu)
[WS-Client][handleHeader] RX: Date: Sat, 28 Sep 2019 11:04:10 GMT
[WS-Client][handleHeader] RX: Connection: upgrade
[WS-Client][handleHeader] RX: Upgrade: websocket
[WS-Client][handleHeader] RX: Sec-WebSocket-Accept: 8/xxxxxxxxxxxxxxxxxxxxxxxxxxx
[WS-Client][handleHeader] RX: Sec-WebSocket-Protocol: arduino
[WS-Client][handleHeader] Header read fin.
[WS-Client][handleHeader] Client settings:
[WS-Client][handleHeader]  - cURL: /
[WS-Client][handleHeader]  - cKey: xxxxxxxxxxxxxxxxxxxxxxxxx
[WS-Client][handleHeader] Server header:
[WS-Client][handleHeader]  - cCode: 101
[WS-Client][handleHeader]  - cIsUpgrade: 1
[WS-Client][handleHeader]  - cIsWebsocket: 1
[WS-Client][handleHeader]  - cAccept: 8/xxxxxxxxxxxxxxxxxxxxxxxxxx
[WS-Client][handleHeader]  - cProtocol: arduino
[WS-Client][handleHeader]  - cExtensions: 
[WS-Client][handleHeader]  - cVersion: 0
[WS-Client][handleHeader]  - cSessionId: 
[WS-Client][handleHeader] Websocket connection init done.
[WS][0][headerDone] Header Handling Done.
[WSc] Service connected to sinric.com at url: /
Waiting for commands from sinric.com ...
```
So I guess that when ESPs use static IPs they don't look up the DNS address through the router's configuration but we have to specify the DNS server to use (that is either the router's IP or a generic DNS IP as Cloudflare's, Google's ...) 

Maybe wrong here but this seem to be the case...