kakopappa/sinric #345: French (CA) translation help needed

kakopappa, 2019-09-30:
Please help me to translate this into French so I can publish

One Sentence Description *
Connect your WeMos mini D1, ESP8266, ESP32 or Arduino development boards with Amazon Alexa or Google Home for free!

Detailed Description *
With Sinric, you can connect your WeMos mini D1, ESP8266, ESP32, Arduino development boards with Alexa without emulating as a Belkin switch.

How to use?

Goto sinric.com

Create an account and log in

Create a new device. eg: tv

Use the API key displayed on the home screen to connect your development board.

Here is an example of how to connect :

https://github.com/kakopappa/sinric/blob/master/arduino_examples/switch_example.ino

replace the "Place your API key here", "your wifi ssid", "your wifi password" in the code with your details

Install / Enable Sinric smart home skill

Ask Alexa to discover devices.

Say Alexa, turn on "tv"

Example Phrases *

Alexa, turn on tv
Alexa, turn on bedroom lights
Alexa, dim bedroom lights

Keywords:
Connect, ESP8266, to, Alexa, connect, Wemos, D1, mini, Alexa, Arduino, Alexa

yoan31, 2019-10-05:
Hi !

Connect your WeMos mini D1, ESP8266, ESP32 or Arduino development boards with Amazon Alexa or Google Home for free!

--> Connectez votre WeMos mini D1, ESP8266, ESP32 ou carte de développement Arduino avec Amazon Alexa ou Google Home gratuitement !

With Sinric, you can connect your WeMos mini D1, ESP8266, ESP32, Arduino development boards with Alexa without emulating as a Belkin switch.
--> Avec Sinric, vous pouvez connecter votre WeMos mini D1, ESP8266, ESP32 ou carte de développement Arduino sans émulation comme interrupteur Belkin.

How to use? --> Comment l'utiliser ?

Goto sinric.com --> Aller sur le site sinric.com

Create an account and log in --> Créer un compte et se connecter.

Create a new device. eg: tv --> Créer un nouveau périphérique ex: tv

Use the API key displayed on the home screen to connect your development board. 
--> Utilisez la clé API affichée sur l'écran d'accueil pour connecter votre carte de développement.

Here is an example of how to connect :
--> Voici un exemple de connexion:

replace the "Place your API key here", "your wifi ssid", "your wifi password" in the code with your details
--> remplacez le "Placez votre clé API ici", "votre nom de wifi", "votre mot de passe wifi" dans le code avec vos infos.

Install / Enable Sinric smart home skill
--> Installer / Activer Sincric sur le skil Sinric smart home

Ask Alexa to discover devices.
--> Demandez à Alexa de découvrir de nouveaux appareils.

Say Alexa, turn on "tv"
--> Dites Alexa, allume la "tv"

Alexa, turn on bedroom lights
--> Dites Alexa, allume les lumières de la chambre

Alexa, dim bedroom lights
--> Dites Alexa, lumières tamisées a la chambre

kakopappa, 2019-10-06:
@yoan31 Thanks

kakopappa, 2019-10-06:
can you help me with the upcoming pro version text as well, please?

yoan31, 2019-10-06:
If I can yes with pleasure

kakopappa, 2019-10-09:
Hi there. Sorry about the delay in getting back to you. 

I ran the English text through Google Translate so it will make your life easy (or worse). it seems to do a pretty good job but sometimes it needs corrections. You can file the translated content here

https://github.com/sinricpro/language-files/tree/master/fr

English text is here:
https://github.com/sinricpro/language-files/tree/master/en

Most important one is https://github.com/sinricpro/language-files/blob/master/fr/amazon_alexa_skill.txt as I plan to submit the skill for approval this week.

Thanks a lot for you help, your help will be rewarded

kakopappa, 2019-10-15:
@yoan31 hi buddy, any update on this?

Can you atlease help me with https://github.com/sinricpro/language-files/blob/master/fr/amazon_alexa_skill.txt ? I will be submitting others languages this tomorrow. I like to add support for French as well.

yoan31, 2019-10-15:
Hi.
Sorry for the delay...
Too many things to do.

I want to correct this directly but I can't on Github.

Some corrections :

*Public Name*
Sinric Pro

*One Sentence Description*
Connectez votre NodeMCU, votre WeMos D1 mini, votre ESP8266, votre ESP32,
votre Raspberry Pi, votre Arduino et d’autres cartes de développement à
Sinric Pro pour ajouter l’intégration Amazon Alexa GRATUITEMENT!

*Detailed Description*
Avec Sinric Pro, vous pouvez connecter votre carte de développement IOT à
Alexa avec une compétence fiable: finie l’émulation glitchy. Sinric Pro
prend en charge tous les types de périphériques Amazon IOT et propose des
bibliothèques Python, C ++ et NodeJS avec des exemples pour vous permettre
de démarrer en quelques minutes. Pour les utilisateurs avancés, vous pouvez
créer vos propres périphériques, tableau de configuration et routines
personnalisées et surveiller tous vos périphériques via l'API REST.
Déverrouillez GRATUITEMENT dès aujourd'hui l'intégration Alexa hautement
disponible, hautement sécurisée et bien documentée pour vos projets!

Commencer:

1. Visiter https://sinric.pro

2. Créer un compte gratuit et connectez-vous

3. Depuis l'application Alexa ou le tableau de bord, installer et activer
la compétence "Sinric Pro" Smart Home

4. Créer un nouveau périphérique, par exemple: TV. Notez le jeton
d'autorisation, la clé de signature et l'ID de votre nouveau périphérique.

5. L'application Amazon Alexa affichera une notification push indiquant
qu'elle a découvert votre nouvel appareil.

7. Télécharger cet exemple de croquis ou utiliser nos bibliothèques Python
ou C ++ existantes sur votre tableau de projet:

https://github.com/kakopappa/sinric/blob/master/arduino_examples/switch_example.ino

Remplacer le jeton, la clé de signature et les informations
d'identification WiFI par les vôtres.

8. C’est tout!

*Exemple de phrases*
"Alexa, allumes la télévision"
"Alexa, éteins la télévision"

Keywords
Alexa, Domotique, Bricolage, IOT, Maison intelligente, NodeMCU, WeMos /
Lolin D1 mini, Systèmes Expressif, ESP8266, ESP32, Arduino, RaspberryPi