kakopappa/sinric #346: Hindi (IN) translation help needed

kakopappa, 2019-09-30:
Please help me to translate this into Hindi so I can publish

One Sentence Description *
Connect your WeMos mini D1, ESP8266, ESP32 or Arduino development boards with Amazon Alexa or Google Home for free!

Detailed Description *
With Sinric, you can connect your WeMos mini D1, ESP8266, ESP32, Arduino development boards with Alexa without emulating as a Belkin switch.

How to use?

Goto sinric.com

Create an account and log in

Create a new device. eg: tv

Use the API key displayed on the home screen to connect your development board.

Here is an example of how to connect :

https://github.com/kakopappa/sinric/blob/master/arduino_examples/switch_example.ino

replace the "Place your API key here", "your wifi ssid", "your wifi password" in the code with your details

Install / Enable Sinric smart home skill

Ask Alexa to discover devices.

Say Alexa, turn on "tv"

Example Phrases *

Alexa, turn on tv
Alexa, turn on bedroom lights
Alexa, dim bedroom lights

Keywords:
Connect, ESP8266, to, Alexa, connect, Wemos, D1, mini, Alexa, Arduino, Alexa

adukale, 2019-10-12:
*One Sentence Description *
_Connect your WeMos mini D1, ESP8266, ESP32 or Arduino development boards with Amazon Alexa or Google Home for free!_

`आपका WeMos mini D1, ESP8266, ESP32 या Arduino डेव्हलोपमेंट बोर्ड  अमेझॉन अलेक्सा और गुगल होम से फ्री मे नियंत्रित करे. `

**Detailed Description ***

_With Sinric, you can connect your WeMos mini D1, ESP8266, ESP32, Arduino development boards with Alexa without emulating as a Belkin switch._  

`Sinric के साथ आप आपका  WeMos mini D1, ESP8266, ESP32, Arduino बोर्ड किसीभी दुसरे प्रकार के अनुकरण के बिना जोड शकते है और नियंत्रित कर शकते है How to use?कैसे उपयोग करे ?Goto sinric.com  sinric.com को जाईये  `

_Create an account and log in_

`आपका अकाउंट तयार और साइन इन करें`

_Create a new device. eg: tv_

`कोईभी नया उपकरण तयार किजीये, जैसेकी TV `

_Use the API key displayed on the home screen to connect your development board._

`आपके अकाउंट के होम स्क्रीन पे दिखाई हुई API key के साहाय्य से डेव्हलोपमेंट बोर्ड से जोडिये`

_Here is an example of how to connect :_

`यहां एक उदाहरण है कि कैसे कनेक्ट किया जाए:`

https://github.com/kakopappa/sinric/blob/master/arduino_examples/switch_example.ino

_replace the "Place your API key here", "your wifi ssid", "your wifi password" in the code with your details_  

`"Place your API key here" के जगह पे आपकी API key , "your wifi ssid" और "your wifi password"  के जगह पे आपक wifi का नाव और पासवर्ड डाले. ` 

_Install / Enable Sinric smart home skill_

`Sinric smart home स्किल  इंस्टॉल करें`

_Ask Alexa to discover devices._

`Alexa को डिवाइस खोजने के लिए कहेंSay Alexa, turn on "tv"अलेक्सा TV चालू करो बोले. 
`
_Example Phrases *_

उदाहरण:

_Alexa, turn on tv_  

`Alexa, TV चालू करो`

_Alexa, turn on bedroom lights_ 

` Alexa, बेडरूम कि लाईट चालू करो `

_Alexa, dim bedroom lights_

`Alexa बेडरूम लाईट मंद करो`

kakopappa, 2019-10-13:
Thanks,

Do you mind checking the upcoming pro version text as well please?

https://github.com/sinricpro/language-files/blob/master/hi/amazon_alexa_skill.txt

I have used google translate but not sure how accurate it is

adukale, 2019-10-13:
[](https://github.com/sinricpro/language-files/blob/master/hi/amazon_alexa_skill.txt)
These are okay but not completely accurate. I will modify the same and post here.

kakopappa, 2019-10-13:
thanks a lot

closed 2019-10-15 by kakopappa