kakopappa/sinric #347: requirement of api for user registration and device creation

iot9637, 2019-10-04:
Dear sir,
sir please give the api details for user registration and device creation in sinric.com

imdhanush, 2019-10-05:
@iot9637 It's available next verison of sinric (sinric pro)

iot9637, 2019-10-05:
can u please send me the link??

<https://www.avast.com/en-in/recommend?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=default3&tag=ef1f549a-b6ad-4d82-9ca0-7404d7ddafe9>
I’m
protected online with Avast Free Antivirus. Get it here — it’s free forever.
<https://www.avast.com/en-in/recommend?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=default3&tag=ef1f549a-b6ad-4d82-9ca0-7404d7ddafe9>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

imdhanush, 2019-10-05:
Sure! It's https://sinric.com 
But don't sign up yet! Wait till it goes live .

iot9637, 2019-10-05:
i have gone through sinricpro API details.But it doesn't contain any method
or request for new user registration.So please update API document
including new user registration.

<https://www.avast.com/en-in/recommend?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=default3&tag=ef1f549a-b6ad-4d82-9ca0-7404d7ddafe9>
I’m
protected online with Avast Free Antivirus. Get it here — it’s free forever.
<https://www.avast.com/en-in/recommend?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=default3&tag=ef1f549a-b6ad-4d82-9ca0-7404d7ddafe9>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

kakopappa, 2019-10-06:
@iot9637 we do not intend to support creating users via API due to people are writing bots / abusing the system. Why do you want to create users?