kakopappa/sinric #351: Thermostat sensor function

marcosmoura91, 2019-10-13:
Hello,
At the moment I am using the thermostat function to control an AC over IR. 

Remote controlling and temperature setting work well but I am wondering, since I found no information about the set server status using dht11. 
I have this sensor and I implemented the functions and I update it but what can I do with it?

Is there a special Alexa command which can reply me the actual temperature? Where to see this? What's the function of this?
In the sinric app there should be an option to send more values and not only on and off. 
Since I have a dht11 and send back to the server the temperature I should also be able to see it in the sinric app or?

I'm struggling to find the propos of sending temperature values back to the sinric server.

kakopappa, 2019-10-13:
https://github.com/kakopappa/sinric/blob/master/arduino_examples/switch_example_wemos_d1_mini_with_relay.ino

marcosmoura91, 2019-10-13:
Hi, I have it working. What I don't understand is what can I do with the temperature feedback?

What is the. Point of sending the ambienttemperature and ambienthumidity back to the server?
Is there an Alexa voice command to retrieve this values?

marcosmoura91, 2019-10-14:
Hello guys, today I was messing around after reading some similar posts about the issue.

At the moment the ambienttemperature is working well and updates on sinric. 
The setpoint I will try tomorrow if I can update it to the server also. 

The two different commands to obtain different values go as follows:

Let's say my thermostat name is AC:
-Alexa what is AC set to? 
Alexa answers the set point temperature

-Alexa what is the AC temperature? 
Alexa replies with the sensor reading updated to the ambienttemperature var.

kakopappa, 2019-10-15:
Hi @marcosmoura91 

Setting the ambienttemperature and set point temperature will be support correctly in upcoming Sinric v2. (will be released at the end of this month). I do not intead to fix issues related v1 at the moment.

https://sinric.pro/

marcosmoura91, 2019-10-15:
It's working. I just didn't know what exactly what was the sentence to ask Alexa so she could answer correctly.

artn721, 2020-03-23:
Hello Aruna (Kakopappa) 

I want thank so much you for your heavy and amazing job to help us to connect to alexa.

I implemented light_example and working perfectly.  
Now I try implement termostat DHT11 and have two problems (I will hidden my number to 5xxxxx in below reports:):

1. when I change COOL TO or HEAT TO temperature on my phone I received : 
[WSc] get text: {"deviceId":"5xxxxxxxxxxxxxxxxxxxxx","action":"SetTargetTemperature","value":{"lowerSetpoint":{"value":42,"scale":"CELSIUS"},"upperSetpoint":{"value":43,"scale":"CELSIUS"}}}
[WSc] SetTargetTemperature value: 
[WSc] SetTargetTemperature scale: 
[WSc] Webservice disconnected from sinric.com!
[WSc] Service connected to sinric.com at url: /
Waiting for commands from sinric.com ...

2. when is changed temperature on DHT I see on terminal :
OK	19.0		24.0		75.2		23.0		73.3
but is not changed Current Temp on my phone application and Alexa also told to me that temperature is all time constant the same what was set on voice command from Alexa:
OK	20.0		22.0		71.6		20.8		69.4
[WSc] get text: {"deviceId":"5xxxxxxxxxxxxxxxxxxxxxx","action":"SetTargetTemperature","value":{"targetSetpoint":{"value":41,"scale":"CELSIUS"}}}
[WSc] SetTargetTemperature value: 41
[WSc] SetTargetTemperature scale: CELSIUS
OK	19.0		24.0		75.2		23.0		73.3 

I enclose file with existing program :)

[thermostat_example_with_dht_11org.zip](https://github.com/kakopappa/sinric/files/4369287/thermostat_example_with_dht_11org.zip)
[image]

Can you help me in this case?

Best Regards 
Artur

artn721, 2020-03-23:
Hello Mr. Aruna,

I solved first problem with this error. I said to Alexa "change thermostat mode to heating" and on phone changed template (enclosed on photo) and I see only HEAT and now after change target I can change both side from application and voice. It working very good. 
[WSc] get text {"deviceId":"5xxxxxxxxxxxxxxxxxxxxxx","action":"SetTargetTemperature","value":{"targetSetpoint":{"value":18,"scale":"CELSIUS"}}}
[WSc] SetTargetTemperature value: 18
[WSc] SetTargetTemperature scale: CELSIUS

But the second problem all time I haven't solve. Maybe I do something or understand wrong. Maybe some people also had similar problem and can help me.

Best regards 
Artur
[image]