kakopappa/sinric #354: forgot option is not available

shubhabanerjeewin8, 2019-10-15:
I can't find forgot option.

kakopappa, 2019-10-17:
post the email here, i will reset it and let you know

akh13, 2019-12-15:
I was hoping to reset my password as well. Is there actually an option to do so yet via the API or on the UI?

imdhanush, 2019-12-15:
@akh13  please upgrade to [Sinric Pro](https://sinric.pro) . It's available there.

kakopappa, 2019-12-17:
@akh13 sorry password reset option is not available atm and there is no API. 

I got two solutions for you

1. Please consider switching to Pro version if you can. It includes password reset, and a lot more features. You have access to 5 devices for free for life

2. You can drop your email here. I will reset the password and email it to you.