kakopappa/sinric #355: Smart lock sinric option

servermcp, 2019-10-16:
Hello @kakopappa do you plan to add the "Smart lock" device to the options?  I am working on one and I want it to work as such and not settle for using as a switch.
 Thank you!!

Hola @kakopappa tienes pensado agregar el dispositivo "Smart lock" a las opciones? Estoy trabajando en una y quisiera que funcione como tal y no conformarme con usar como interruptor.
Gracias!!

kakopappa, 2019-10-17:
Hi,

Are you using Alexa or Google Home?

servermcp, 2019-10-17:
Google Home!

kakopappa, 2019-10-17:
Well, This will be fixed in v2 release. However, might take sometime. Reason is I like to get the Google Home action approved and in the store this time. Last time (with Sinric) when we tried to get the action approved, Google rejected us because they didn't like "the way" we ask users to enable it (creating your own action), plus  we have to send two devices for their testing center in USA for approval. This is a long process.

Short answer is buy a Alexa. It will be available after the Pro skill get published in couple of days.

bakliwalp, 2020-03-15:
I am using Alexa,can you share the code for smart lock using Sinric

RinkuNezert, 2020-07-20:
I have the same problem with "speacker", "Fan", "SmartLock" and "Windows Shutter" they appear as a catch on the google home application

RinkuNezert, 2020-07-20:
I have the same problem with "speaker", "Fan", "Smart Lock" and "Windows Shutter", they appear as an electrical outlet on the google home app