kakopappa/sinric #356: 433 Mhz RF Garage Door Project

Federicobj, 2019-10-17:
Hi @kakopappa, I am working on a simple project and I need to see if you can help me.
I have a garage door that opens with a 433mhz rf signal and what I want to do is using a nodemcu esp8266 + a 433mhz rf emitter that when giving an alexa "command" the rf signal is sent to the gate and I don't know how to implement it in the arduino sketch, any help will be very grateful, Federico

kakopappa, 2019-10-17:
Not sure how good you are with Arduino programming. If I was you I would
split this into 3 tasks

1. Wire the ESP module + RF and send a command. confirm that you can open /
close the garage door using ESP module

2. Alexa - Esp link. Create a lock device in Sinric. Watch commands appear
in the Serial terminal as you ask Alexa to lock / unlock the door (check
the steps on home page). This is simple.

Use this Arduino sketch
https://github.com/kakopappa/sinric/blob/master/arduino_examples/switch_example.ino

3. Merge the sketches

Good luck

Federicobj, 2019-10-17:
@kakopappa, my arduino programing is not very good :( , I generay download samples and modify it, I never program any sketch from the scratch
1- I just clone the RF signal of my garage with an rf receiver, and then I send it with an emitter with two simple sketchs, thats works OK
2- I also made a sinric commando to turn on or off a light, that works great also :)
3- what I dont know how to do is to join this two things :):):), I think someting like this will work, but i think that someting is wrong on the sketch couse I do not see were to put the rf code, protocol, etc.
https://www.instructables.com/id/Amazon-Echo-Switches-433mHz-Remote-Outlets-With-ES/

By the way, thanks for the super fast responce, and if you have any advice on how to do it or what is wrong on this sketh it will be very appreciated

kakopappa, 2019-10-17:
Upload the sketch here without the API key. Me or someone else will help
you when have time

Federicobj, 2019-10-17:
Thaks you very much for yout advice :):):) i get it to work !!!...My only doubt is which device in the list of sinric "device type" has only a "power toggle" button and not on/off state buton? besides that the sketch works great for open and close my garage my edited parts of your switch sketch are in BOLD, here you have the sketh if someone needs it

here is the sketch:
/*
 Version 0.4 - April 26 2019
*/ 
**#include <RCSwitch.h>**
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <WebSocketsClient.h> //  https://github.com/kakopappa/sinric/wiki/How-to-add-dependency-libraries
#include <ArduinoJson.h> // https://github.com/kakopappa/sinric/wiki/How-to-add-dependency-libraries (use the correct version)
#include <StreamString.h>

ESP8266WiFiMulti WiFiMulti;
WebSocketsClient webSocket;
WiFiClient client;

**RCSwitch mySwitch = RCSwitch();**

#define MyApiKey "deleted" // TODO: Change to your sinric API Key. Your API Key is displayed on sinric.com dashboard
#define MySSID "deleted" // TODO: Change to your Wifi network SSID
#define MyWifiPassword "deleted" // TODO: Change to your Wifi network password

#define HEARTBEAT_INTERVAL 300000 // 5 Minutes 

uint64_t heartbeatTimestamp = 0;
bool isConnected = false;

// deviceId is the ID assgined to your smart-home-device in sinric.com dashboard. Copy it from dashboard and paste it here

void turnOn(String deviceId) {
  if (deviceId == "deleted") // Device ID of first device
  {  
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    **mySwitch.send(154703253, 28);
    delay(1000);**
  }     
}

void turnOff(String deviceId) {
   if (deviceId == "deleted") // Device ID of first device
   {  
     Serial.print("Turn off Device ID: ");
     Serial.println(deviceId);
     **mySwitch.send(154703253, 28);
     delay(1000);**
   }
}

void webSocketEvent(WStype_t type, uint8_t * payload, size_t length) {
  switch(type) {
    case WStype_DISCONNECTED:
      isConnected = false;    
      Serial.printf("[WSc] Webservice disconnected from sinric.com!\n");
      break;
    case WStype_CONNECTED: {
      isConnected = true;
      Serial.printf("[WSc] Service connected to sinric.com at url: %s\n", payload);
      Serial.printf("Waiting for commands from sinric.com ...\n");        
      }
      break;
    case WStype_TEXT: {
        Serial.printf("[WSc] get text: %s\n", payload);
        // Example payloads

        // For Switch or Light device types
        // {"deviceId": xxxx, "action": "setPowerState", value: "ON"} // https://developer.amazon.com/docs/device-apis/alexa-powercontroller.html

        // For Light device type
        // Look at the light example in github
          
        DynamicJsonBuffer jsonBuffer;
        JsonObject& json = jsonBuffer.parseObject((char*)payload); 
        String deviceId = json ["deviceId"];     
        String action = json ["action"];
        
        if(action == "setPowerState") { // Switch or Light
            String value = json ["value"];
            if(value == "ON") {
                turnOn(deviceId);
            } else {
                turnOff(deviceId);
            }
        }
        else if (action == "SetTargetTemperature") {
            String deviceId = json ["deviceId"];     
            String action = json ["action"];
            String value = json ["value"];
        }
        else if (action == "test") {
            Serial.println("[WSc] received test command from sinric.com");
        }
      }
      break;
    case WStype_BIN:
      Serial.printf("[WSc] get binary length: %u\n", length);
      break;
  }
}

void setup() {
  **Serial.begin(9600);**
  
  **// Transmitter is connected to Arduino Pin D4
  mySwitch.enableTransmit(2);**
  
  **// Optional set protocol (default is 1, will work for most outlets)
  mySwitch.setProtocol(6);**
  
  WiFiMulti.addAP(MySSID, MyWifiPassword);
  Serial.println();
  Serial.print("Connecting to Wifi: ");
  Serial.println(MySSID);  

  // Waiting for Wifi connect
  while(WiFiMulti.run() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  if(WiFiMulti.run() == WL_CONNECTED) {
    Serial.println("");
    Serial.print("WiFi connected. ");
    Serial.print("IP address: ");
    Serial.println(WiFi.localIP());
  }

  // server address, port and URL
  webSocket.begin("iot.sinric.com", 80, "/");

  // event handler
  webSocket.onEvent(webSocketEvent);
  webSocket.setAuthorization("apikey", MyApiKey);
  
  // try again every 5000ms if connection has failed
  webSocket.setReconnectInterval(5000);   // If you see 'class WebSocketsClient' has no member named 'setReconnectInterval' error update arduinoWebSockets
}

void loop() {
  webSocket.loop();
  
  if(isConnected) {
      uint64_t now = millis();
      
      // Send heartbeat in order to avoid disconnections during ISP resetting IPs over night. Thanks @MacSass
      if((now - heartbeatTimestamp) > HEARTBEAT_INTERVAL) {
          heartbeatTimestamp = now;
          webSocket.sendTXT("H");          
      }
  }   
}

// If you want a push button: https://github.com/kakopappa/sinric/blob/master/arduino_examples/switch_with_push_button.ino

kakopappa, 2019-10-18:
awsome.