kakopappa/sinric #357: Thermostat vs. Google Home

gborus, 2019-10-19:
Hi @kakopappa ,

first of all many thanks for the effort you put into this great and fun project!

I'm trying to put together a Thermostat control for Google Home.
I was using the example you posted for Alexa. Now I can see the thermostat device in Google Home app, however it seems the communication is only one directional, it works from Google Home to the ESP device. This means that I can see the commands in the serial monitor which the ESP receives.

My question is how can I update the server with data from the thermostat. e.g. send the actual temperature or tell Google Home the thermostat mode?
I found your example for Alexa for this: setTargetTemperatureOnServer, but I'm not sure how I should change it to work with Google Home. 
Based on the supported traits doc: https://developers.google.com/assistant/smarthome/traits/temperaturesetting.html#response it seems that Google expects much more data from the device and also, it seems that I should pass back the 'requestId' value in my response to google from ESP, however I didn't find the requestId from Sinric.
Also I though that Google Home will send "action.devices.SYNC" or "action.devices.QUERY" actions to gather data from the thermostat, however that' not received after ESP connects to Sinric, nor when I ask the temperature from Google.

For me it would be a great help if you could share a sample code for e.g. updating the temperature on the server or how to send the current settings from the ESP to Google. From there I'll try to implement the other actions.

Thanks,
-gabor.

kakopappa, 2019-10-19:
Update the temperature

https://github.com/kakopappa/sinric/blob/master/arduino_examples/thermostat_example_with_dht_11.ino

Don’t overload sending in a loop()

https://github.com/kakopappa/sinric/blob/master/arduino_examples/how_to_update_server_example.ino

gborus, 2019-10-19:
Hi @kakopappa ,

Thanks for the quick turnaround.
It is working.

QQ: is SetTemperatureSetting (root["action"] = "SetTemperatureSetting";) your own action for Sinric communication?
I didn't spot this action in google's documentation...
I'd just like to understand how these actions are related to Google Thermostat device's actions.

thanks
-gabor.