kakopappa/sinric #358: 'DynamicJsonBuffe error

stanboyd, 2019-10-20:
Here is error message, how do I get 'DynamicJsonBuffe fixed please.

Arduino: 1.8.10 (Windows 10), Board: "WeMos D1 R2 & mini, 80 MHz, 921600, 4M (3M SPIFFS)"

C:\Users\joeha\OneDrive\Documents\Arduino\sketch_sinricrelay\sketch_sinricrelay.ino: In function 'void webSocketEvent(WStype_t, uint8_t*, size_t)':

sketch_sinricrelay:78:9: error: 'DynamicJsonBuffer' was not declared in this scope

         DynamicJsonBuffer jsonBuffer;

         ^

sketch_sinricrelay:78:27: error: expected ';' before 'jsonBuffer'

         DynamicJsonBuffer jsonBuffer;

                           ^

sketch_sinricrelay:79:28: error: 'jsonBuffer' was not declared in this scope

         JsonObject& json = jsonBuffer.parseObject((char*)payload); 

                            ^

C:\Users\joeha\OneDrive\Documents\Arduino\sketch_sinricrelay\sketch_sinricrelay.ino: In function 'void setPowerStateOnServer(String, String)':

sketch_sinricrelay:161:3: error: 'DynamicJsonBuffer' was not declared in this scope

   DynamicJsonBuffer jsonBuffer;

   ^

sketch_sinricrelay:161:21: error: expected ';' before 'jsonBuffer'

   DynamicJsonBuffer jsonBuffer;

                     ^

sketch_sinricrelay:162:22: error: 'jsonBuffer' was not declared in this scope

   JsonObject& root = jsonBuffer.createObject();

                      ^

sketch_sinricrelay:167:8: error: 'class ArduinoJson660_0_0::JsonObject' has no member named 'printTo'

   root.printTo(databuf);

        ^

C:\Users\joeha\OneDrive\Documents\Arduino\sketch_sinricrelay\sketch_sinricrelay.ino: In function 'void setTargetTemperatureOnServer(String, String, String)':

sketch_sinricrelay:176:3: error: 'DynamicJsonBuffer' was not declared in this scope

   DynamicJsonBuffer jsonBuffer;

   ^

sketch_sinricrelay:176:21: error: expected ';' before 'jsonBuffer'

   DynamicJsonBuffer jsonBuffer;

                     ^

sketch_sinricrelay:177:22: error: 'jsonBuffer' was not declared in this scope

   JsonObject& root = jsonBuffer.createObject();

                      ^

sketch_sinricrelay:181:57: error: invalid initialization of non-const reference of type 'ArduinoJson660_0_0::JsonObject&' from an rvalue of type 'ArduinoJson660_0_0::JsonObject'

   JsonObject& valueObj = root.createNestedObject("value");

                                                         ^

sketch_sinricrelay:182:76: error: invalid initialization of non-const reference of type 'ArduinoJson660_0_0::JsonObject&' from an rvalue of type 'ArduinoJson660_0_0::JsonObject'

   JsonObject& targetSetpoint = valueObj.createNestedObject("targetSetpoint");

                                                                            ^

sketch_sinricrelay:187:8: error: 'class ArduinoJson660_0_0::JsonObject' has no member named 'printTo'

   root.printTo(databuf);

        ^

Multiple libraries were found for "ESP8266WiFi.h"
 Used: C:\Users\joeha\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi
Multiple libraries were found for "WebSocketsClient.h"
 Used: C:\Users\joeha\Documents\Arduino\libraries\arduinoWebSockets-2.1.4
Multiple libraries were found for "ArduinoJson.h"
 Used: C:\Users\joeha\Documents\Arduino\libraries\ArduinoJson
 Not used: C:\Users\joeha\Documents\Arduino\libraries\ArduinoJson-5.13.2
Multiple libraries were found for "Hash.h"
 Used: C:\Users\joeha\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\Hash
exit status 1
'DynamicJsonBuffer' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

sivar2311, 2019-11-05:
Code is correct, but made for ArduinoJson Version 5 while ArduinoJson Version 6 is installed.
Quick solution: Downgrade to ArduinoJson Version 5.
Long solution: Make code compatible to Version 6 see [Migrating from version 5 to 6](https://arduinojson.org/v6/doc/upgrade/)

sivar2311, 2019-11-28:
Hi @stanboyd !
I forked this repo and started to make the examples compatible to ArduinoJson 5 & 6.
https://github.com/sivar2311/sinric/tree/master/arduino_examples
Actually all switch examples compiling fine on both ArduinoJson versions, but i am not able to test them. Do you have the time for testing the switch examples? 
If they're working, i will do a pull request to this repo.