kakopappa/sinric #360: Is there any way for me to help add the brazilian portuguese language

CaptainNoodle, 2019-10-22:
Hi,

I woul like to know if there is any way for me to help add the brazilian portuguese language?

Best regards.

CaptainNoodle, 2019-10-22:
Sorry, I didn't that there was a request for pt-br already.

Regards!

ebnack, 2020-02-27:
Você conseguiu confirgurar?

CaptainNoodle, 2020-02-27:
Sim, consegui e estou usando.

ebnack, 2020-02-27:
Postei agora a pouco uma dúvida sobre o código..
não sei se está certo... 
Poderias ver se é isso mesmo, estou em dúvidas no código.

https://github.com/kakopappa/sinric/issues/394

CaptainNoodle, 2020-02-27:
Aparentemente, sim. O que acontece exatamente quando vc envia o código ao ESP?

ebnack, 2020-02-27:
Aparece esse erro
- na parte de /*comandos para desligar*/
-- /*11 - D10 - GPIO 1*/
erro: **'else' without a previous 'if'**

CaptainNoodle, 2020-02-27:
Com quantos devices você está trabalhando? Se você tiver apenas um device não precisa de todos aqueles que são definidos no começo do código.

Tem algum ";' em lugar errado. Vi que vc colocou vários em resposta ao erro

error: expected ';' before '{' token

Isso não vai resolver o seu problema.

Você está utilizando a versão mais atualizada dos exemplos?

Pode ser que a versão de exemplo que vc está usando esteja com algum bug.

Pode explicar exatamente o que vc quer fazer com esse código?

ebnack, 2020-02-27:
1 módulo esp8266
estou na versão mais recente.

Quero, com cada pino (GPIO), controlar uma lampada em casa, ligar e desligar pelo google home.
no código do portão, a idéia é que ele envie um pulso de 3 segundos, e desligue, para acionar o motor para abrir ou fechar o portão.

CaptainNoodle, 2020-02-27:
Recomendo você começar com apenas um dispositivo e ir aumentando um a um à
medida que for instalando mais dispositivos para verificar se o código está
ok.

ebnack, 2020-02-27:
Farei isso!
Abraço