kakopappa/sinric #364: Error while compiling example

ilkayefe, 2019-10-28:
Hi, while im trying to upload the google home switch example to my nodemcu i get these errors.
How can i fix this problem? I have pasted error messages that i got below.

```
c:/users/handan/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: cannot open output file C:\Users\handan\AppData\Local\Temp\arduino_build_240757/Google_Home_Switch_Sinric.ino.elf: Permission denied

collect2.exe: error: ld returned 1 exit status

Multiple libraries were found for "ESP8266WiFi.h"
 Used: C:\Users\handan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi
Multiple libraries were found for "WebSocketsClient.h"
 Used: C:\Users\handan\Documents\Arduino\libraries\arduinoWebSockets-2.1.4
Multiple libraries were found for "ArduinoJson.h"
 Used: C:\Users\handan\Documents\Arduino\libraries\ArduinoJson
Multiple libraries were found for "Hash.h"
 Used: C:\Users\handan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\Hash
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).
```

kakopappa, 2019-10-29:
Multiple libraries were found for "Hash.h"

Looks like you have two libraries which is using the same "Hash.h" causing this issue

ilkayefe, 2019-10-29:
Thank you so much! I have found the second Hash.h file and deleted. It fixed my issue.

closed 2019-10-29 by ilkayefe