kakopappa/sinric #365: IRShine

IacoT1, 2019-10-30:
Hello @kakopappa,
many compliments for your elegant and effective work.
I would like to know what do you think about my project: https://github.com/IacoT1/IRShine
Do you think it would be possible to combine the projects to obtain a very powerful, all purposes, universal remote. Your suggestions are welcome.

kakopappa, 2019-10-30:
wow looks pretty neat. love it.. how much does it cost to produce ?

I have Sinric V2 (Pro) coming out soon and I love to promote small makers (like you) work accessible to everyone via Sinric platform.  Sinric will provide a comprehensive app, api and documentation which requires to integrate your hardware with the platform. Right now, I am waiting for the skill to be approved and planing to go for beta testing somewhere next week . Take a look at the site, app and the api and let me know what you think?   https://sinric.pro 

Thanks

IacoT1, 2019-10-30:
I think that Sinric Pro is the right and open way to create new products and solutions by merging hardware with your software platform. Again my compliments for your initiative.
I'm actually using IRShine with Tasmota, MQTT and Home Assistant. The hardware is incredibly powerful: you can place an irshine even hidden behind a furniture and it controls everything in every room. The Oslon emitter is really a monster...
Actually i have little control over the production costs because i have produced only few pieces but with very good quality in mind. The considerations are:
- the oslon is probably the most expensive part but it's also the strong point of the project
- the power supply (Meanwell) is a top quality but can be replaced by a less expensive model (IRM-10 it's pin compatible but you can use any no-brand...)
- the trimmer R3 is present for those who want to try to overdrive the emitter but it's not necessary and, by placing a correct value for fixed resistor R2, can be shorted
- the pcb production is very low cost by using a service like JLCPCB
- the IR cap is 3D printed using PETG filament but there are many services in China that can produce low cost molded parts.
My personal weakness is the software because of my hardware formation.
If you allow me, i will send you my email address, let me know...
Many thanks

kakopappa, 2019-10-30:
of course.