kakopappa/sinric #366: Thermostat is not working with DHT.

ilkayefe, 2019-10-30:
Hi, i uploaded thermosat example with dht to my nodemcu, but dht is not giving values. Im using google home app. My log stays like this after 1 minutes. I have tried the test code of "DHTesp.h" library and thats worked with the same wiring.
```
Connecting to Wifi: SUPERONLINE_WiFi_0909
.....
WiFi connected. IP address: 192.168.1
[WSc] Service connected to sinric.com at url: /
Waiting for commands from sinric.com ...
[WSc] get text: {"deviceId":"5db6badea08667203520bf51","action":"action.devices.commands.ThermostatTemperatureSetpoint","value":{"thermostatTemperatureSetpoint":27.5}}
```
Also how can we use these commands in google home? Sorry for my questions, im new to C language.

[image]

kakopappa, 2019-10-31:
can you post your sketch here without the API key?

kakopappa, 2019-10-31:
If you have issues reading values from DHT sensor, then i like to recommend you to create a simple sketch to do that and make sure it works first, then you can integrate that code to GH sketch

ilkayefe, 2019-10-31:
> If you have issues reading values from DHT sensor, then i like to recommend you to create a simple sketch to do that and make sure it works first, then you can integrate that code to GH sketch

I tried example of "DHTesp" library, that worked. Im using this code -> https://github.com/kakopappa/sinric/blob/master/arduino_examples/thermostat_example_with_dht_11.ino

ilkayefe, 2019-11-02:
How i can fix this issue?

murilobaliego, 2020-05-20:
Hi @kakopappa, how can I send the temperature value that I get on my DHT sensor to GH server? I tried to send for instance the value "33.3" like below but it did not worked.
[WSc] sendTXT: {"deviceId":"5ec40b1fef8d6c2a9102cd11","action":"action.devices.commands.ThermostatTemperatureAmbient","value":{"thermostatTemperatureAmbient":{"value":"33.3"}}}

Am I using the wrong action commad?
Thanks in advance!! I'm looking forward to use GH in sinric PRO :) congrats for this great work on sinric!!

closed 2021-03-01 by ilkayefe