kakopappa/sinric #367: Google home often says: Sorry, it looks like Sinric is unavailable right now.

Deimos1994, 2019-11-05:
It happens multiple times a day for 10 - 30 seconds then it works again.
WebSocket still stays connected, so no problems here.
This behavior started maybe a week ago or so.
Should I re-login with Sinric in the Google Home app?
Is it only me who has this issue?
Is the server ok?

kakopappa, 2019-11-05:
@Deimos1994  hey. it's been a while. 

Last night i checked the server see whether anyone one is flooding the server and I did not see anyone. I think the server is processing a lot of requests per second now and may be failing to respond to GH request on time. 

Last couple of months I have been working with few other guys to building a new server, app and a SDK from scratch to solve issues like this. Guys did a greate job with the SDKs. The server can horizontally scale well and It's in beta now (https://sinric.pro) however it supports only Alexa. For GH, you know what happened last time. This time we will approcah this problem bit differently. I will submit for approval without letting users create their DIY action. I am planning to complete the GH integration in a week or two and then submit for action approval + sending two modules to US for testing. I think this will take atleast 1 month. Till then i will try to find quick workaround to "Sinric is unavailable right now"  problem. Sorry about this.

kakopappa, 2019-11-05:
duh, i think i forgot to renew the letsencrypt SSL cerificates again

jackallen28, 2019-11-05:
So you will be fixing this issue soon? PS. love your work :) Great app, amazing stuff!

kakopappa, 2019-11-05:
yes. I am still at work..... have to do this when i go back home. will be in couple of hours from now :)

kakopappa, 2019-11-05:
The issue was not related to SSL, it was due to load. Too many connections. 

I am using Nginx server reverse proxy the connections to an internal nodejs server. Nginx was randomly throwing this error to log

 *35005 2048 worker_connections are not enough while connecting to upstream

So i increased the no of connections. Should be stable now

jackallen28, 2019-11-05:
It is working much better now. No issues so far. Thank you for your help!

Deimos1994, 2019-11-05:
@kakopappa, Ok, it works now, thanks for fixing. Closing.

closed 2019-11-05 by Deimos1994