kakopappa/sinric #369: Sinric Pro is here

kakopappa, 2019-11-14:
Sinric Pro is here for beta testing. 
https://sinric.pro

Please give it a try and let us know how you feel. Thanks. 💯

lsgdefault, 2019-11-23:
what's difference between sinric and sinric pro??

kakopappa, 2019-11-23:
- Sinric Pro supports 13 device types such as contact sensors / motion sensors / doorbells 
check out the doorbell demo: https://www.youtube.com/watch?v=H-iftzWVTXE

[image]

- New website with a new dashboard / create new devices/ rooms / audit logs / create credentials

[image]

- Improved security. Every message is authenticated, signed and must be verified. 

-

Gerardmoik, 2019-12-05:
Is Sinric going to be replaced by Sinric pro at any point? or we will be able to use Sinric forever? thanks.

kakopappa, 2019-12-05:
Yes, it will be replaced. There is an operating cost involved in running
sinric.com

Developer-error, 2019-12-28:
Sinric pro is good but sinric is faster then sinric pro

sivar2311, 2019-12-28:
What do you mean with "faster"?
Did you do any measurements?
I have two light devices (1x Sinric, 1x SinricPro):
Turn on/off command Sinric: about 1.5 seconds
Turn on/off command SinricPro: about 1 second

kakopappa, 2019-12-29:
@Developer-error Could be but there are few things here you may not know.

1. Pro has additional checks for security like HMAC signature checks. These may delay response processing

2. Sinric - When you ask Alexa to do something, it forwards the request to Sinric server and the Sinric server returns OK response first whether the device is connected or not. Whereas Sinric pro sends the command to ESP module, wait for the response and send it back to Alexa. This creates a small delay

Developer-error, 2019-12-29:
Got it sir. 
I m trying to learn  from sinric , Big fan of your's.

daniele-castellini, 2020-01-02:
> Yes, it will be replaced. There is an operating cost involved in running sinric.com
> […](#)

When will it be discontinued?

Gerardmoik, 2020-01-02:
I had a paid subscription at Smartnest.cz but I moved to Sinric.com because it is free, it is a pity that Sinric.com cannot stay forever but I understand there must be some costs for it. Now Sinric Pro sounds interesting, I will give it a try. is there any expected date for Sinric.com to be shut down?

kakopappa, 2020-01-02:
@Gerardmoik @daniele-castellini I do not have an exact date as of now. We are waiting for the Google Home action to be approved so that existing users can switch to pro seamlessly. I will send an email for at least 3 months in advance with the date. 

I understand that you guys must be disappointed. I get it. I like to keep it free forever too. but I like you guys to see this is a screencap from Alexa skill dev console. As you can see the Sinric Alexa Skil makes about 3000-7000 requests per hour just Alexa only!! At this rate, I have to sell a kidney to keep this up and running... Please consider switching to Pro if you can. 

[image]

gians-it, 2020-01-07:
I'm checking  your  good work, for now everything's perfect!
may I make a suggestion:
in SinricProSwitch   an example with 2 devices would be appreciated.

thanks

sivar2311, 2020-01-07:
> in SinricProContactsensor an example with 2 devices would be appreciated.

You can have as much devices as you want.
Just call `<SinricProDeviceType> &<NAME> = SinricPro[<DEVICEID>]` so it would be like this:
```
SinricProContactsensor &Sensor1 = SinricPro["5e14e91b9764a6c6d709b802"];
SinricProContactsensor &Sensor2 = SinricPro["5e14e93816266f609c211df4"];
SinricProContactsensor &Sensor2 = SinricPro["5e14e96d030cf63ef98ffa0b"];
..
SinricProContactsensor &Sensor99 = SinricPro["5e14e975e82097a9276508c0"];
```

You can do this with an array of DeviceIDs and iterate over that array to setup the devices.

Have a look at [this example](https://github.com/sinricpro/esp8266-esp32-sdk/issues/19#issuecomment-571128657) where i am using a std::map to setup multiple switch devices. The map contains as first parameter the deviceId, and as second parameter a struct which stores the used PINs, and the actual power state).

You can do this technique with all SinricPro devices in the library. Not only switches and contact sensors.

gians-it, 2020-01-08:
thank you very much for your very rapid response!
I think two devices are the best solution.
for example using a cheap ESP-01

while I'm trying  occasionally it informs me:
`[SinricPro]: EVENTS ARE BLOCKED FOR 58 SECONDS!`
I probably have to check what I wrote and change a few things   :-(
or use a or use a _**debounce capacitor**_. 
Regards

sivar2311, 2020-01-08:
A simple debounce logic will help (see [Arduino debounce example](https://www.arduino.cc/en/tutorial/debounce))

Events are limited to prevent flooding the server (like you would have done ;).
Current limitation is 10 events with a minimum distance of 1 second, then events are blocked for 1 minute.

gians-it, 2020-01-10:
Hi, 
in SinricProContactsensor

can I use **INPUT_PULLUP** instead of:

 ` pinMode(CONTACT_PIN, INPUT);`

sivar2311, 2020-01-10:
That depends on the contact sensor you're using.
And keep in mind to change the detection logic! In the example, an event will be send, when PIN is HIGH. Using INPUT_PULLUP, the pin will be always HIGH, except it is pulled to GND by our contact sensor.
What type of contact sensor do you use?

gians-it, 2020-01-10:
I'm not sure between Hall magnetic sensor or optocoupler.
I'm just working on it,  the software side works great by the SinricPro aid.
 thank you so much for your help.

temocastro, 2020-01-30:
@kakopappa when will Sinric stop working? Alexa Discover Devices is not working for new devices I just created

aguilaair, 2020-02-06:
@kakopappa any updates regarding Google Home integration?  Also: will we be able to choose which action we would like to enable (by this I mean that in V1 even if a light can only be dimmed the GH app would still allow choosing colours) ? Thank you in advace!

Gerardmoik, 2020-02-06:
Also looking forward to the google action, for now my only option is Smartnest.cz but it uses mqtt instead of web sockets. When will it be released? Thank you !

kakopappa, 2020-02-06:
[image]

Waiting for approval 💯

Gerardmoik, 2020-02-06:
Great news !!! Thanks !

aguilaair, 2020-02-06:
Thank you! 
[image]
How long have you been waiting?

robademar, 2020-02-10:
since I've to change my sinric working code I would like to have a more secure usage. I've found that sometime commands are not received by devices, I don't know if it is due to networking ore other reasons.
How can I have a feedback?
Then, since Alexa allows user voice recognition, I would like to check who is giving the command. Is it possible with the pro version? Thanks

sivar2311, 2020-02-10:
>  I've found that sometime commands are not received by devices, I don't know if it is due to networking ore other reasons.

Did you have this problem with Sinric (classic) or SinricPro?

> Then, since Alexa allows user voice recognition, I would like to check who is giving the command. Is it possible with the pro version.

SinricPro <--> Alexa connection is based on Alexa SmartHome Skill Interface. Unfortunately this Interface does not support user voice recogition. SinricPro does not get any voice informations. Maybe amazon will provide such info's on SmartHome interface one day...we have to wait for.

> Since I want a reliable service this is not possible for free, but it should be reasonable (2-3 $/year)

Then SinricPro meets your requirement with 5 devices for free and additional devices are $3 per year.

robademar, 2020-02-10:
I've found that sometime commands are not received by devices, I don't know if it is due to networking ore other reasons.

Did you have this problem with Sinric (classic) or SinricPro?   SINric classic Just tonigth it was not working for a couple of hours then without any change it worked

Then, since Alexa allows user voice recognition, I would like to check who is giving the command. Is it possible with the pro version.

SinricPro <--> Alexa connection is based on Alexa SmartHome Skill Interface. Unfortunately this Interface does not support user voice recogition. SinricPro does not get any voice informations. Maybe amazon will provide such info's on SmartHome interface one day...we have to wait for.Let's hope

Since I want a reliable service this is not possible for free, but it should be reasonable (2-3 $/year)

Then SinricPro meets your requirement with 5 devices for free and additional devices are $3 per year.We are aligned  Thanks
Giovanni Roberto Mariani

sivar2311, 2020-02-11:
> I've found that sometime commands are not received by devices, I don't know if it is due to networking ore other reasons.

Sinric (classic) had no issue that time. So this will be related to WiFi issues and might happen with SinricPro and other WiFi related projects, too.

robademar, 2020-02-11:
Clear thanks. And it will be like this till a more reliable 2 ways protocol will be adopted and command feed  back transmitted.
Inviato da Yahoo Mail su Android

kakopappa, 2020-02-11:
@robademar 

When Alexa sends a request to the Sinric (sinric.com) the server sends "Success" back to Alexa without checking whether device is connected or not

In Sinric Pro, when Alexa sends a command to Sinric Pro (sinric.pro) server, it forwards the command to ESP module and wait for 8 seconds for a response from the ESP module. If your device is not connected or does not respond back to server Alexa will say device is not responding. If you hear device is not responding often likely you have wifi issues either with the router or esp board

pacjo, 2020-02-26:
When google home support? Dashboard says: "Pending approval" for 3 weeks.

kakopappa, 2020-02-26:
@pacjo I wish it was simple as Alexa. After we submitted the action for approval, after two weeks we got an email from Google asking to ship 2 devices for testing to Google in USA.  We are preparing the devices for testing and will ship to Google soon. Then don't know how long it will take to approval.

Banjak88, 2020-03-06:
Hello sirs,
I have a problem with doorbell at sinricpro. After creating account and creating a new doorbell device and after preparing nodemcu and uploading code it shows online on the dashboard and when pressing the bell icon alexa plays the doorbell sound but when pressing the flash pushbotton on nodemcu nothing occured note that I can see the event on serial monitor. So what is the problem do you think (Nb: when pressing Turn on it shows waiting then it won't turn on.)

sivar2311, 2020-03-06:
Wrong repository ;)

Please enable debug and post the serial log.

Banjak88, 2020-03-06:
sivar2311 thanks for the fast reply and sorry about commenting about that here.
Anyway this is the debug serial output:

22:43:38.936 
22:43:38.937 [Wifi]: Connecting.connected!
22:43:39.197 [WiFi]: IP‑Address is 192.168.1.152
22:43:39.236 [SinricPro:add()]: Adding device with id "5e5cfed486622d498dd52453".
22:43:39.308 [SinricPro:Websocket]: Connecting to WebSocket Server (ws.sinric.pro)
22:43:39.383 [SinricPro:Websocket]: headers: 
22:43:39.419 appkey:80bcce84‑cabd‑4143‑b88c‑9e41963bd8ac
22:43:39.465 deviceids:5e5cfed486622d498dd52453
22:43:39.503 restoredevicestates:false
22:43:39.531 platform:ESP8266
22:43:39.549 version:2.4.0
22:43:40.953 [SinricPro:Websocket]: connected
22:43:40.989 Connected to SinricPro
22:43:41.013 [SinricPro:Websocket]: receiving data
22:43:41.055 [SinricPro.handleReceiveQueue()]: 1 message(s) in receiveQueue
22:43:41.121 [SinricPro.handleReceiveQueue()]: Signature is valid. Processing message...
22:43:41.202 [SinricPro:extractTimestamp(): Got Timestamp 1583527422
22:43:58.871 [SinricPro:Websocket]: disconnected
22:43:58.907 Disconnected from SinricPro
22:43:59.401 [SinricPro:Websocket]: connected
22:43:59.437 Connected to SinricPro
22:44:00.413 [SinricPro:Websocket]: receiving data
22:44:00.453 [SinricPro.handleReceiveQueue()]: 1 message(s) in receiveQueue
22:44:00.520 [SinricPro.handleReceiveQueue()]: Signature is valid. Processing message...
22:44:00.600 [SinricPro:extractTimestamp(): Got Timestamp 1583527441
22:44:01.564 Ding dong...
22:44:01.579 [SinricPro:sendMessage()]: pushing message into sendQueue
22:44:01.640 [SinricPro:handleSendQueue()]: 1 message(s) in sendQueue
22:44:01.699 [SinricPro:handleSendQueue()]: Sending message...
22:44:01.752 {
22:44:01.755   "header": {
22:44:01.772     "payloadVersion": 2,
22:44:01.799     "signatureVersion": 1
22:44:01.828   },
22:44:01.834   "payload": {
22:44:01.850     "action": "DoorbellPress",
22:44:01.883     "cause": {
22:44:01.900       "type": "PHYSICAL_INTERACTION"
22:44:01.939     },
22:44:01.947     "createdAt": 1583527442,
22:44:01.979     "deviceId": "5e5cfed486622d498dd52453",
22:44:02.026     "replyToken": "69397514‑ae15‑4393‑abe0‑03a970dccc15",
22:44:02.088     "type": "event",
22:44:02.111     "value": {
22:44:02.126       "state": "pressed"
22:44:02.154     }
22:44:02.161   },
22:44:02.169   "signature": {
22:44:02.186     "HMAC": "oh0iJcgEEFQZWzvj7o7S512KxLp/7gWoH/FuBBonKH8="
22:44:02.249   }
22:44:02.255 }
22:44:02.258 [SinricPro:handleSendQueue]: Sending to websocket
22:44:02.310 [SinricPro:handleSendQueue()]: message sent.
22:44:09.405 [SinricPro:Websocket]: disconnected
22:44:09.443 Disconnected from SinricPro
22:44:10.873 [SinricPro:Websocket]: connected
22:44:10.908 Connected to SinricPro
22:44:10.933 [SinricPro:Websocket]: receiving data
22:44:10.974 [SinricPro.handleReceiveQueue()]: 1 message(s) in receiveQueue
22:44:11.041 [SinricPro.handleReceiveQueue()]: Signature is valid. Processing message...
22:44:11.121 [SinricPro:extractTimestamp(): Got Timestamp 1583527451

sivar2311, 2020-03-07:
>22:43:58.871 [SinricPro:Websocket]: disconnected
>22:43:58.907 Disconnected from SinricPro
>22:43:59.401 [SinricPro:Websocket]: connected

Disconnect no 1 

>22:44:02.258 [SinricPro:handleSendQueue]: Sending to websocket
>22:44:02.310 [SinricPro:handleSendQueue()]: message sent.
>22:44:09.405 [SinricPro:Websocket]: disconnected
>22:44:10.873 [SinricPro:Websocket]: connected

Disconnect no 2 (right after, or while sending the event message)

It seems you'r ESP have an unstable wifi connection and the event message might not send completely. 
Does this happen always? Please try again and check the serial log for dissconnections like this.

Banjak88, 2020-03-07:
This happens always, before and after pressing the push button. (note: the dashboard shows repeatedly online message in green and shows the doorbell device power off when trying to power it on by pressing Turn on botton it shows waiting then power off again).
Also the esp harware itself was tested before and its working properly and i have another esp runs sinric tv ir and working well on the same network.

sivar2311, 2020-03-07:
Right now, doorbell devices doesn't support on/off (dashboard is wrong here). And i guess your sketch doesnt implement onPowerState, right? ;) So forget about on/off button in dashboard -> this have to be fixed (removing on/off button in dashboard).

Did you use the original doorbell example sketch, or are there any modifications? If there are any modifications, please post your sketch (and circuit diagram for external connected circuits).

The original sketch doesn't disconnect itself.

We have to find out, what is causing the disconnects...this is very difficult because it can happen everywhere in your configuration (weak wifi signal, router, esp module issues, external circuit issues  etc etc).

Banjak88, 2020-03-07:
Yes I had used the original example code without modifications and yes on the original code there is no onPowerState,
There is no exta circuit its nodemcu with buildin pushbotton labeled "flash" connected to gpio 0,
I think WebSocketClient library is used for connection to sinricpro server what is the tested library version I will check for that meanwhile. Again thanks for cooperation.

sivar2311, 2020-03-07:
Ok, please let me know if you found out something

sivar2311, 2020-03-07:
edit: doorbell device can support onPowerState, but it's not implemented in example sketch.
Sorry, i got confused about this.

So it is possible to check the powerstate by implementing onPowerState callback and turn on / off the doorbell.

Banjak88, 2020-03-07:
Ok, I will test implementing onPowerState.
I noticed that the sinricpro site shows at the upper right corner sinricpro version (v2.2.3), but my arduino library is (v2.4.0). Is it make sence if i downgrade my arduino library to (v2.2.3)?

kakopappa, 2020-03-07:
Sinric Pro portal version number is different from SDK version number.

Use the latest version of SDK always

Banjak88, 2020-03-07:
I had implemented onPowerState to doorbell code that found on the switch example.
take a look please did I messed anything?

`/*
 * Example for how to use SinricPro Doorbell device:
 * - setup a doorbell device
 * - send event to sinricPro server if button is pressed
 * 
 * If you encounter any issues:
 * - check the readme.md at https://github.com/sinricpro/esp8266-esp32-sdk/blob/master/README.md
 * - ensure all dependent libraries are installed
 *   - see https://github.com/sinricpro/esp8266-esp32-sdk/blob/master/README.md#arduinoide
 *   - see https://github.com/sinricpro/esp8266-esp32-sdk/blob/master/README.md#dependencies
 * - open serial monitor and check whats happening
 * - check full user documentation at https://sinricpro.github.io/esp8266-esp32-sdk
 * - visit https://github.com/sinricpro/esp8266-esp32-sdk/issues and check for existing issues or open a new one
 */

// Uncomment the following line to enable serial debug output
#define ENABLE_DEBUG

#ifdef ENABLE_DEBUG
       #define DEBUG_ESP_PORT Serial
       #define NODEBUG_WEBSOCKETS
       #define NDEBUG
#endif 

#include <Arduino.h>
#ifdef ESP8266 
       #include <ESP8266WiFi.h>
#endif 
#ifdef ESP32   
       #include <WiFi.h>
#endif

#include "SinricPro.h"
#include "SinricProDoorbell.h"

#define WIFI_SSID         "Banjak"
#define WIFI_PASS         "talabanjak645"

//#define WIFI_SSID         "banjaky7"
//#define WIFI_PASS         "banjak.181"

#define APP_KEY           "80bcce84-cabd-4143-b88c-9e41963bd8ac"      // Should look like "de0bxxxx-1x3x-4x3x-ax2x-5dabxxxxxxxx"
#define APP_SECRET        "3c564271-3568-49c4-ad46-d226c94bdf1a-fe698f6c-8549-4307-87a8-f186f8dc43c0"   // Should look like "5f36xxxx-x3x7-4x3x-xexe-e86724a9xxxx-4c4axxxx-3x3x-x5xe-x9x3-333d65xxxxxx"
#define DOORBELL_ID       "5e5cfed486622d498dd52453"    // Should look like "5dc1564130xxxxxxxxxxxxxx"

#define BAUD_RATE         9600                // Change baudrate to your need

// change this to your button PIN
// on NodeMCU D3 / GPIO-0 is flash button PIN so you can use the builtin flash button
#define BUTTON_PIN 0
#define POWER_BUTTON_PIN 16   // on NodeMCU D0 / GPIO-16
#define LED_PIN   2   // GPIO for middle BiuldInLed (inverted)

bool myPowerState = false;
unsigned long lastBtnPress = 0;

/* bool onPowerState(String deviceId, bool &state) 
 *
 * Callback for setPowerState request
 * parameters
 *  String deviceId (r)
 *    contains deviceId (useful if this callback used by multiple devices)
 *  bool &state (r/w)
 *    contains the requested state (true:on / false:off)
 *    must return the new state
 * 
 * return
 *  true if request should be marked as handled correctly / false if not
 */
bool onPowerState(const String &deviceId, bool &state) {
  Serial.printf("Device %s turned %s (via SinricPro) \r\n", deviceId.c_str(), state?"on":"off");
  myPowerState = state;
  digitalWrite(LED_BUILTIN, myPowerState?LOW:HIGH);
  return true; // request handled properly
}

// checkButtonpress
// reads if BUTTON_PIN gets LOW and send Event
void checkButtonPress() {
//  static unsigned long lastBtnPress;
  unsigned long actualMillis = millis();

  if (actualMillis-lastBtnPress > 500) {
    if (digitalRead(BUTTON_PIN)==LOW) {
      Serial.printf("Ding dong...\r\n");
      lastBtnPress = actualMillis;

      // get Doorbell device back
      SinricProDoorbell& myDoorbell = SinricPro[DOORBELL_ID];
      myDoorbell.onPowerState(onPowerState);
      // send doorbell event
      myDoorbell.sendDoorbellEvent();
    }
  }
    actualMillis = millis();
    if (digitalRead(BUTTON_PIN) == LOW && actualMillis - lastBtnPress > 500)  { // is button pressed (inverted logic! button pressed = LOW) and debounced?
    if (myPowerState) {     // flip myPowerState: if it was true, set it to false, vice versa
      myPowerState = false;
    } else {
      myPowerState = true;
    }
    digitalWrite(LED_PIN, myPowerState?LOW:HIGH); // if myPowerState indicates device turned on: turn on led (builtin led uses inverted logic: LOW = LED ON / HIGH = LED OFF)

    // get Doorbell device back
    SinricProDoorbell& myDoorbell = SinricPro[DOORBELL_ID];
    // send powerstate event
    myDoorbell.sendPowerStateEvent(myPowerState); // send the new powerState to SinricPro server
    Serial.printf("Device %s turned %s (manually via D0 button)\r\n", myDoorbell.getDeviceId(), myPowerState?"on":"off");
    }
}

// setup function for WiFi connection
void setupWiFi() {
  Serial.printf("\r\n[Wifi]: Connecting");
  WiFi.begin(WIFI_SSID, WIFI_PASS);

  while (WiFi.status() != WL_CONNECTED) {
    Serial.printf(".");
    delay(250);
  }
  IPAddress localIP = WiFi.localIP();
  Serial.printf("connected!\r\n[WiFi]: IP-Address is %d.%d.%d.%d\r\n", localIP[0], localIP[1], localIP[2], localIP[3]);
}

// setup function for SinricPro
void setupSinricPro() {
  // add doorbell device to SinricPro
  SinricPro.add<SinricProDoorbell>(DOORBELL_ID);
  // setup SinricPro
  SinricPro.onConnected([](){ Serial.printf("Connected to SinricPro\r\n"); }); 
  SinricPro.onDisconnected([](){ Serial.printf("Disconnected from SinricPro\r\n"); });
  SinricPro.begin(APP_KEY, APP_SECRET);
}

// main setup function
void setup() {
  pinMode(BUTTON_PIN, INPUT_PULLUP); // BUTTIN_PIN as INPUT
  pinMode(POWER_BUTTON_PIN, INPUT_PULLUP); // BUTTIN_PIN as INPUT

  Serial.begin(BAUD_RATE); Serial.printf("\r\n\r\n");
  setupWiFi();
  setupSinricPro();
}

void loop() {
  checkButtonPress();
  SinricPro.handle();
}`

sivar2311, 2020-03-07:
```
// setup function for SinricPro
void setupSinricPro() {
  SinricProDoorbell &myDoorbell = SinricPro[DOORBELL_ID];
  myDoorbell.onPowerState(onPowerState);

  SinricPro.onConnected({ Serial.printf("Connected to SinricPro\r\n"); });
  SinricPro.onDisconnected({ Serial.printf("Disconnected from SinricPro\r\n"); });
  SinricPro.begin(APP_KEY, APP_SECRET);
}
```

Banjak88, 2020-03-07:
Ok, I will test that.
thank you.

sivar2311, 2020-03-07:
sorry, for that short response....i am in the middle of a bathroom renovation

Banjak88, 2020-03-07:
> sorry, for that short response....i am in the middle of a bathroom renovation

No problem thank you for your time.

Banjak88, 2020-03-08:
Hello again, my problem solved by adding onPowerState to the original code and adding extra if statement to myPowerState variable to not send events when doorbell device when its turned off by sinricpro.
thanks for cooperation again.

kakopappa, 2020-03-09:
@Banjak88 please open an issue if you have problems in sinricpro repo https://github.com/sinricpro/esp8266-esp32-sdk/issues

Banjak88, 2020-03-09:
> @Banjak88 please open an issue if you have any further issue in sinricpro repo https://github.com/sinricpro/esp8266-esp32-sdk/issues

Ok thanks.

daniel-paiva-osf, 2020-03-25:
I hope we get good news from google soon. I don't want to start on the sinric.com just to change it again to pro after a few moments :)
Keep up the good work!  @kakopappa

buddasticks, 2020-04-09:
Hello I'm having trouble trying to add more devices to the script any input would help.cheers ( might result in a beer )

zendes, 2020-05-10:
Any update on the Google Home integration? It's been pending for 4 months now, which I'm guessing must be frustrating to you. Anything you can share about what's going on with Google?

laxmru, 2020-06-07:
I am not programer. But I make few things using internet. In this month I have spent lot of time to create voice control TV remote using sinric.com. 
I will look forward for voice control IR remote with GH detailed example in sinric pro because I have Google Home mini.
Thanks

laxmru, 2020-06-07:
> I am not programer. But I make few things using internet. In this month I have spent lot of time to create voice control TV remote using sinric.com.
> I will look forward for voice control IR remote with GH detailed example in sinric pro because I have Google Home mini.
> Thanks

Glen1986, 2020-06-18:
how can i connect sinric pro white google home whith no ifttt, like sinric.......
its possible?
thanks a lot

samratnarvekar, 2020-06-29:
> @Gerardmoik @daniele-castellini I do not have an exact date as of now. We are waiting for the Google Home action to be approved so that existing users can switch to pro seamlessly. I will send an email for at least 3 months in advance with the date.
> 
> I understand that you guys must be disappointed. I get it. I like to keep it free forever too. but I like you guys to see this is a screencap from Alexa skill dev console. As you can see the Sinric Alexa Skill makes about 3000-7000 requests per hour just Alexa Alexa Skill only!! At this rate, I have to sell a kidney to keep this up and running... Please consider switching to Pro if you can.
> 
> [image]

But Sir, i have an opinion, if u start charging then i would say i better opt out from ioT and buy readymade smart devices from Sonoff, belkin etc as they provide one time hardware cost and life long skill for free.

samratnarvekar, 2020-06-29:
Hence think well before taking such a step because only project related ioT people will subscribe for sinric pro, or make atleast 10 devices free in pro.

kakopappa, 2020-06-29:
@samratnarvekar you should opt-out

[17 more replies]