kakopappa/sinric #372: Not working properly

lsgdefault, 2019-11-23:
hey so first thing first i would like to thanks developers for making service like sinric for free it help me very much and working stable for past 2-3 months. but yesterday it stopped working i mean alexa was responding okay turned on light and all but it didn't turned on anything so i connected it with my laptop and it gives status like disconnected with sinric web services. for 5-6 hours it was disconnected. after some times i came back to home and checked it so it says connected to web services but now the problem is it gives delay to output and sometime it don't even gives output. i said turn off light and alexa said ok and after 48 seconds it turned off my light so it's not stable now. I want to know it you are closing service because of sinric pro or you are moving server or something. please help me.

kakopappa, 2019-11-23:
Hi @lsgdefault  

Disconnects are hard to pinpoint where it's coming from. It could be the bad code, ESP, or it could be the router / network provider/ the server. Normally WebSocket library reconnects afaik when disconnection was detected. (disconnection detection normally takes time as well). Make sure  you have enabled setReconnectInterval  https://github.com/Links2004/arduinoWebSockets/blob/master/examples/esp8266/WebSocketClient/WebSocketClient.ino#L94 and you should do some test on your own and fix these kinds of issues

Sinric is not shutting down soon... however, It's running on a cheap VPS server and resources are limited at the moment. If someone else is flooding the server with on/off requests it will slow down for the rest of the people. There are few issues like this, that's why I created Pro. The server needs to be rewritten with these problems in mind. Please consider moving to Pro. I will update your other question regarding Sinric vs pro

lsgdefault, 2019-11-23:
As i said everything is perfect in my code and hardware and that code, hardware and home network worked perfectly for past few months so why it is being problem for me now?
And second sinric pro don't give 8 device for free so that's something.

sivar2311, 2019-11-23:
I am running a light on Sinric (non Pro) and have no such issues.
Voice commands are handled within 2 seconds without delay.
So mabye there are some wifi problems that causing this issue!?

lsgdefault, 2019-11-23:
i don't know i didn't change my broadband or something....!!

lsgdefault, 2019-11-23:
This is my code if you want to have a look..
ESP8266 with 8 switch relay
`/*
 Version 0.4 - April 26 2019
*/ 

#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <WebSocketsClient.h> //  https://github.com/kakopappa/sinric/wiki/How-to-add-dependency-libraries
#include <ArduinoJson.h> // https://github.com/kakopappa/sinric/wiki/How-to-add-dependency-libraries (use the correct version)
#include <StreamString.h>

int device_1 = 5;
int device_2 = 4;
int device_3 = 0;
int device_4 = 2;
int device_5 = 14;
int device_6 = 12;
int device_7 = 13;
int device_8 = 15;

ESP8266WiFiMulti WiFiMulti;
WebSocketsClient webSocket;
WiFiClient client;

#define MyApiKey "" // TODO: Change to your sinric API Key. Your API Key is displayed on sinric.com dashboard
#define MySSID "Deco Ash" // TODO: Change to your Wifi network SSID
#define MyWifiPassword "123123123" // TODO: Change to your Wifi network password

#define HEARTBEAT_INTERVAL 300000 // 5 Minutes 

uint64_t heartbeatTimestamp = 0;
bool isConnected = false;

// deviceId is the ID assgined to your smart-home-device in sinric.com dashboard. Copy it from dashboard and paste it here

void turnOn(String deviceId) {
  if (deviceId == "") // Device ID of first device
  {  
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    digitalWrite(device_1, LOW);
  } 
  else if (deviceId == "") // Device ID of second device
  { 
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    digitalWrite(device_2, LOW);
  }
  else if (deviceId == "") // Device ID of Third device
  { 
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    digitalWrite(device_3, LOW);
  }
  else if (deviceId == "") // Device ID of Fourth device
  { 
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    digitalWrite(device_4, LOW);
  }
  else if (deviceId == "") // Device ID of Fifth device
  { 
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    digitalWrite(device_5, LOW);
  }
  else if (deviceId == "") // Device ID of Sixth device
  { 
[... 164 lines cut ...]
  // server address, port and URL
  webSocket.begin("iot.sinric.com", 80, "/");

  // event handler
  webSocket.onEvent(webSocketEvent);
  webSocket.setAuthorization("apikey", MyApiKey);
  
  // try again every 5000ms if connection has failed
  webSocket.setReconnectInterval(5000);   // If you see 'class WebSocketsClient' has no member named 'setReconnectInterval' error update arduinoWebSockets
}

void loop() {
  webSocket.loop();
  
  if(isConnected) {
      uint64_t now = millis();
      
      // Send heartbeat in order to avoid disconnections during ISP resetting IPs over night. Thanks @MacSass
      if((now - heartbeatTimestamp) > HEARTBEAT_INTERVAL) {
          heartbeatTimestamp = now;
          webSocket.sendTXT("H");          
      }
  }   
}
`

lsgdefault, 2019-11-24:
Suddenly started working don't know what but now working perfectly thanks for your support

closed 2019-11-24 by lsgdefault