kakopappa/sinric #375: SinricPro vs Sinric - json

steve8266, 2019-11-26:
Kakopappa,
I have used the sinric example code extensively over recent months and thank you very much indeed for that (its turning on/off loads of stuff in my home!). Back then I soon found that I needed json5 to get it to compile. In the last week or so I've been getting started with sinricPro, initially the doorbell example. I'm looking  forward to using more of Pro and with some 14 devices I'll happily pay for the 9 or so over the free 5!
However I cant get the doorbell to compile without going to json6 which means I cant compile the original sinric stuff without swapping json versions backwards and forwards.  So I've tried to migrate the old stuff to json6 using the migration instructions at https://arduinojson.org/v6/doc/upgrade/. Trouble is its too complex for me as a novice...

So, a request please. Unless I'm doing something really dumb(?) I cant be the only one out here with this issue and I'd very much appreciate some advice, or better still example code, to uplift the original sinric switch example/s to use json6?

Your help please.
Regards, Steve
PS. realise SinricPro is work in progress but noticed clicking on the arduino example code at https://sinric.pro/index.html - 'get started now' step4 ESP8266/ESP32 Example goes to a 404 error.

kakopappa, 2019-11-26:
Thanks for your input

Solution 1:

I like to recommend you to switch to PlatfromIO + VS Code IDE from Arduino. Platform IO offers a much more feature-rich IDE for ESP8266/ESP32 developers.

steve8266, 2019-11-26:
Many thanks for your fast reply. I just had a look at PlatformIO for the first time and it is impressive (especially the context editor/s). Its a bit complex though, for me at least, but I will start work getting to know it in order to move across.  

In the meantime if its not too much trouble I'd like Solution2 aswell please... I'm using the 'arduino_examples/switch_example.ino'  plus the setPowerStateOnServer procedure, all of it for Alexa only.  I'm not using any push switches or light type devices (or temperature stuff). I'd be happy to do the testing if it saves you time. 
thanks again, Steve

sivar2311, 2019-11-27:
Hi @steve8266 !
I forked this repo made **all switch examples** compatible to ArduinoJson version **5 and 6**.
All switch examples should compile fine on both versions now. 
But the examples have to be tested before merging them.
Unfortunately i'am not able to test them. Would you please test **all switch examples** and give us a feedback?
Link to my repo is https://github.com/sivar2311/sinric/tree/master/arduino_examples

steve8266, 2019-11-27:
Hi Sivar,
That’s great, thanks. I’ll happily test the switches. It will take me some days though as I’m away from home at the moment. Regards, Steve

steve8266, 2019-11-30:
Hi Sivar,
I've tested all the switch_examples on both json5 and json6 and they all work fine. Also I tested the 'switch_with_push_button' example to check that setPowerStateOnServer is working - it is.
All of them work fine thank you. Just to convince myself it wasn't just fluke, I've also confirmed that the code runs thru the #if ARDUINOJSON_VERSION_MAJOR == 5(/6)  statements as it is supposed to.  Test conditions:
Arduino 1.8.10 using board NodeMCU 1.0
ArduinoJson by Benoit Blanchon vs 5.13.4, for J5 tests (obviously).
ArduinoJson by Benoit Blanchon vs 6.13.0 for J6

Note please I couldn't do the google_home_switch because don't have that device.

Thanks again for your help with this.
Regards, Steve

sivar2311, 2019-12-01:
Hi steve,
thank you for your testing!

My repo has been merged -> Examples are now compatible to both versions of ArduinoJson.

closed 2019-12-04 by steve8266