kakopappa/sinric #378: FASTLED + Sinric

riotm0de, 2019-12-12:
Hello,

can someone help me or post me a sketch, 
to get an ESP8266 with Fastled connected to Sinric/Alexa ? 

THANK YOU!!!

sivar2311, 2019-12-12:
What is your exact question / problem?

riotm0de, 2019-12-12:
hi @sivar2311, thanks for reply!

my Wemos Mini has FastLED on it and i want to control it with Alexa. Did some research and found a YT Video, who has done this with Sinric. can you give me some advise how to do so?  
or send me a sketch ? 
Thank you!

sivar2311, 2019-12-12:
1st advice is to us SinricPro and SinricPro library available via ArduinoIDE Library Manager or github (https://github.com/sinricpro/esp8266-esp32-sdk).
Light example will be in next release, but you can use the example code from dev-2.2.7 branch (https://github.com/sinricpro/esp8266-esp32-sdk/blob/dev-2.2.7/examples/Light/Light.ino)

2nd advice is to use FastLED library which is also available via ArduinoIDE Library Manager and well documented on Github: https://github.com/FastLED/FastLED

3rd advice: Have a look at ESP8266 notes about FastLED (https://github.com/FastLED/FastLED/wiki/ESP8266-notes) (if you want to control a led stripe). Otherwise you will have flickering issues.

Question: What's your code right now? What did you try, what libraries do you use? What's your ESP8266 / C++ / Arduino knowledges?