kakopappa/sinric #383: Default mode OFF

sivateja369, 2019-12-21:
Hai Kankopappa .....

How are you doing.

Everything works fine for me. But i install nudemcu in my main board. I don’t have power backup.on  power failure the nodemcu turn on and relay also goes to High.

Can you kindly help on this case

Thanks

sivar2311, 2019-12-21:
You might switch over to [SinricPro](https://sinric.pro).
There is a feature [`SinricPro.restoreDeviceStates(true)`](https://sinricpro.github.io/esp8266-esp32-sdk/class_sinric_pro_class.html#a27a9bb7f1e8bde0f39398649493b6f93) exactly for this situation (power failure).
See [ESP8266 / ESP32 SDK](https://github.com/sinricpro/esp8266-esp32-sdk)
Library is available via Github, ArduinoIDE Library Manager and PlatformIO Libraries

sivateja369, 2019-12-21:
thanks sivar2311 I will try it

sivateja369, 2020-01-04:
Hello 

I am SCIVTEJ,

I run orphanage home in my place and I’m having 100 to 500 switches which it takes hours to turn them off when all students leave the home so I need to build my own sinric like server and Alexa skill to donate them can you please help me on that

Thanks