kakopappa/sinric #389: No longer work with core 2.6.2

Microsyl, 2020-01-21:
Hi,
    My ESP8266 was working just fine under core 2.5.2 but after upgrading to 2.6.2 the library
arduinoWebSockets-2.1.4 no longer work lot's of error at compilation time

WebSockets.h:33: In file included from
Wifi_Alexa.ino:7: from
 
IPAddress.h: 82:9: error: 'IPAddress::IPAddress(u32_t)' cannot be overloaded
   IPAddress(u32_t address) { ctor32(address); }
 
IPAddress.h: 81:9: error: with 'IPAddress::IPAddress(long unsigned int)
   IPAddress(uint32_t address) { ctor32(address); }
 
IPAddress.h: 113:14: error: 'bool IPAddress::operator==(u32_t) const' cannot be overloaded
   bool operator==(u32_t addr) const {
 
IPAddress.h: 110:14: error: with 'bool IPAddress::operator==(long unsigned int) const
   bool operator==(uint32_t addr) const {
 
IPAddress.h: 119:14: error: 'bool IPAddress::operator!=(u32_t) const' cannot be overloaded
   bool operator!=(u32_t addr) const {
 
IPAddress.h: 116:14: error: with 'bool IPAddress::operator!=(long unsigned int) const
   bool operator!=(uint32_t addr) const {
 Build failed for project 'Wifi_Alexa'
 
core_esp8266_features.h: 89:1: error: unknown type name 'uint32_t
   inline uint32_t esp_get_cycle_count() __attribute__((always_inline))
 
core_esp8266_features.h: 90:1: error: unknown type name 'uint32_t
   inline uint32_t esp_get_cycle_count() {
 
core_esp8266_features.h: In function esp_get_cycle_count
 
core_esp8266_features.h: 91:3: error: unknown type name 'uint32_t
   uint32_t ccount
core_esp8266_features.h: At top level
 
core_esp8266_features.h: 113:24: error: unknown type name 'uint32_t
   void precache(void *f, uint32_t bytes)

Thanks for your time
Sylvain Bissonnette

sivar2311, 2020-01-21:
unknown type name 'uint32_t was a bug in esp core 2.6.1 and was fixed in 2.6.2. So i am confused having this error in 2.6.2?! For websocket, maybe using another or updated websocket library may help! None of the reported issues are related to sinric. Did you open issues in related repositories?

Microsyl, 2020-01-21:
Ok, I will check for a upgraded websocket

Thanks for your time
Sylvain Bissonnette

closed 2020-01-21 by Microsyl

sivar2311, 2020-01-21:
Maybe give a try on using SinricPro

sivar2311, 2020-01-22:
Upgrade to 2.6.3 is a vailable since December 16th 2019.
The Fix you're looking for is implemented in this version.