kakopappa/sinric #390: Alexa is not discovering

rpinoncely, 2020-01-28:
Hi, I did everything like instructed but Alexa is not discovering my device.

Device is conected to Sinric and the test buttons seems to work, I already enable the skill in Alexa but I still cannot discover it

aukris, 2020-01-28:
I am facing the same issue. Test buttons seem to be working (I can see a serial log) but Alexa can' t discover my ESP8266.

kakopappa, 2020-01-29:
please check now

aukris, 2020-01-29:
Still no luck, There are barely any logs. Can you please tell me how to debug this from my end? Like how does the discovery process work?

kakopappa, 2020-01-30:
OK I will check this when I get back home from work. 

Meantime you should check out https://sinric.pro

temocastro, 2020-01-30:
Same here, not discovering

kakopappa, 2020-01-30:
please check again

temocastro, 2020-01-30:
Ok now

rpinoncely, 2020-02-03:
I worked now, thanks

closed 2020-02-03 by rpinoncely