kakopappa/sinric #395: ESP8266 wifi not working with my alexa command or in app button

gabrieldalbani4, 2020-02-29:
I am using the new code combined with my information, it used to work before the new update to us json 6. Did I do something wrong? It upload fine but intended functions don't work.

/*
 Version 0.4 - April 26 2019
*/ 

#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <WebSocketsClient.h> //  https://github.com/kakopappa/sinric/wiki/How-to-add-dependency-libraries
#include <ArduinoJson.h> // https://github.com/kakopappa/sinric/wiki/How-to-add-dependency-libraries (use the correct version)
#include <StreamString.h>

int device_1 = 03;
int device_2 = 01;
int device_3 = 16;
int device_4 = 05;
int device_5 = 04;
int device_6 = 14;
int device_7 = 12;
int device_8 = 13;

ESP8266WiFiMulti WiFiMulti;
WebSocketsClient webSocket;
WiFiClient client;

#define MyApiKey "8844b4a0-87a1-4dc0-9139-84f175e0e627 " // TODO: Change to your sinric API Key. Your API Key is displayed on sinric.com dashboard
#define MySSID "Dalbani_IoT" // TODO: Change to your Wifi network SSID
#define MyWifiPassword "1000000001" // TODO: Change to your Wifi network password

#define HEARTBEAT_INTERVAL 300000 // 5 Minutes 

uint64_t heartbeatTimestamp = 0;
bool isConnected = false;

// deviceId is the ID assgined to your smart-home-device in sinric.com dashboard. Copy it from dashboard and paste it here

void turnOn(String deviceId) {
  if (deviceId == "5d530d48ea25201d63c93282") // Device ID of first device
  {  
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    digitalWrite(device_1, LOW);
  } 
  else if (deviceId == "5d530d5bea25201d63c93287") // Device ID of second device
  { 
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    digitalWrite(device_2, LOW);
  }
  else if (deviceId == "5d530d6dea25201d63c9328f") // Device ID of third device
  { 
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    digitalWrite(device_3, LOW);
  }
  else if (deviceId == "5d530d93ea25201d63c9329d") // Device ID of fourth device
  { 
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    digitalWrite(device_4, LOW);
  }
  else if (deviceId == "5d530ad4ea25201d63c931bf") // Device ID of fourth device
  { 
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    digitalWrite(device_5, LOW);
  }
  else if (deviceId == "5d530ae6ea25201d63c931c7") // Device ID of fourth device
  { 
[... 224 lines cut ...]
#endif        
  root["action"] = "SetTargetTemperature";
  root["deviceId"] = deviceId;

#if ARDUINOJSON_VERSION_MAJOR == 5
  JsonObject& valueObj = root.createNestedObject("value");
  JsonObject& targetSetpoint = valueObj.createNestedObject("targetSetpoint");
#endif
#if ARDUINOJSON_VERSION_MAJOR == 6        
  JsonObject valueObj = root.createNestedObject("value");
  JsonObject targetSetpoint = valueObj.createNestedObject("targetSetpoint");
#endif  
  targetSetpoint["value"] = value;
  targetSetpoint["scale"] = scale;
   
  StreamString databuf;
#if ARDUINOJSON_VERSION_MAJOR == 5
  root.printTo(databuf);
#endif
#if ARDUINOJSON_VERSION_MAJOR == 6        
  serializeJson(root, databuf);
#endif  
  
  webSocket.sendTXT(databuf);
}

kakopappa, 2020-03-01:
@gabrieldalbani4, code looks fine. 

cannot connect?

gabrieldalbani4, 2020-03-02:
No, through the serial monitor it says it is connected. When I click on or off in the Alexa app the relay doesn't do anything at all. Neither with this code nor blynks code I tried both just to make sure.

chanemi, 2021-05-05:
This is exactly what is happening to me and I have been trying to resolve it for quite some time. The device shows as connected to the WiFi through the serial monitor, also it shows in the Alexa App and the Sinric Pro App but there is no real communication between the Sinric Server and the device. Is there any port forwarding that we need to provision in the router?