kakopappa/sinric #403: user not found

Moriuks, 2020-03-21:
i already downloaded the code and set up my credentials and know im trying to activate the alexa skill but when i put my sincric credentials on the alexa app it says user not found, can any one tell me why this is happening?

Note: My password is solid,even tried this with a new sinric account and still the same issue,im not sure if its because the language

kakopappa, 2020-03-22:
Make sure to use the right skill. There are two

Sinric
Sinric Pro

You should consider using Sinric Pro (https://sinric.pro)

Moriuks, 2020-03-22:
Thanks, yeap the one i setup is Sinric Pro but the mexican Alexa skill store only shows Sinrinc

kakopappa, 2020-03-22:
We have https://www.amazon.es/dp/B07ZT5VDT8 it's not mx but es. I will take a look at mx

I think you have to switch to US to use the Sinric Pro skill for now.

kakopappa, 2020-03-22:
OK, I haven't submitted the MX skill. Can you help us to translate to Spanish (MX) ?

Moriuks, 2020-03-22:
Sure

kakopappa, 2020-03-23:
Thank you for your help. 
 
Alexa Skill page text in English is here:
https://github.com/sinricpro/language-files/blob/master/en/amazon_alexa_skill.txt

you can always reach us on support@sinric.com